osrf/vrx #200: IMU no longer publishing to topic

kartikye, 2020-05-08:
The imu is no longer published when ` <arg name="vrx_sensors_enabled" default="true" />` is in launch file (station_keeping.launch) or when `<arg name="imu_enabled"          default="true" />` is in the launch file (sandisland.launch).

This the output of launching sandisland.

```
logging to /home/kartikye_mittal/.ros/log/2732e324-9103-11ea-9441-42010a800004/roslaunch-instance-2-23790.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://instance-2:40813/
SUMMARY
========
PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True
 * /wamv/robot_description: <?xml version="1....
NODES
  /
    gazebo (gazebo_ros/gzserver)
    spawn_model (gazebo_ros/spawn_model)
auto-starting new master
process[master]: started with pid [23803]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 2732e324-9103-11ea-9441-42010a800004
process[rosout-1]: started with pid [23814]
started core service [/rosout]
process[gazebo-2]: started with pid [23817]
process[spawn_model-3]: started with pid [23825]
[ INFO] [1588925333.028472623]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1588925333.030678508]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised
, waiting...
Warning [parser.cc:950] XML Element[selfCollide], child of element[model] not defined in SDF. Ignoring[selfColli
de]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[static], child of element[link] not defined in SDF. Ignoring[static]. You ma
y have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[static], child of element[link] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[static], child of element[link] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such file or directory
[ INFO] [1588925336.083231322]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1588925336.107887471]: Physics dynamic reconfigure ready.
[ INFO] [1588925336.574179626, 0.194000000]: Thruster namespace <wamv/>
[spawn_model-3] process has finished cleanly
```

This is the call to `rostopic list` for sandisland:
```
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg
/vrx/debug/wind/direction
/vrx/debug/wind/speed
/wamv/joint_states
/wamv/thrusters/left_thrust_angle
/wamv/thrusters/left_thrust_cmd
/wamv/thrusters/right_thrust_angle
/wamv/thrusters/right_thrust_cmd
```

This is the output for station keeping:
```
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg
/vrx/debug/contact
/vrx/debug/wind/direction
/vrx/debug/wind/speed
/vrx/station_keeping/goal
/vrx/station_keeping/pose_error
/vrx/station_keeping/rms_error
/vrx/task/info
/wamv/joint_states
/wamv/sensors/cameras/front_left_camera/camera_info
/wamv/sensors/cameras/front_left_camera/image_raw
/wamv/sensors/cameras/front_left_camera/image_raw/compressed
/wamv/sensors/cameras/front_left_camera/image_raw/compressed/parameter_descriptions
/wamv/sensors/cameras/front_left_camera/image_raw/compressed/parameter_updates
/wamv/sensors/cameras/front_left_camera/image_raw/compressedDepth
/wamv/sensors/cameras/front_left_camera/image_raw/compressedDepth/parameter_descriptions
/wamv/sensors/cameras/front_left_camera/image_raw/compressedDepth/parameter_updates
/wamv/sensors/cameras/front_left_camera/image_raw/theora
/wamv/sensors/cameras/front_left_camera/image_raw/theora/parameter_descriptions
/wamv/sensors/cameras/front_left_camera/image_raw/theora/parameter_updates
/wamv/sensors/cameras/front_left_camera/parameter_descriptions
/wamv/sensors/cameras/front_left_camera/parameter_updates
/wamv/sensors/cameras/front_right_camera/camera_info
/wamv/sensors/cameras/front_right_camera/image_raw
/wamv/sensors/cameras/front_right_camera/image_raw/compressed
/wamv/sensors/cameras/front_right_camera/image_raw/compressed/parameter_descriptions
/wamv/sensors/cameras/front_right_camera/image_raw/compressed/parameter_updates
/wamv/sensors/cameras/front_right_camera/image_raw/compressedDepth
/wamv/sensors/cameras/front_right_camera/image_raw/compressedDepth/parameter_descriptions
/wamv/sensors/cameras/front_right_camera/image_raw/compressedDepth/parameter_updates
/wamv/sensors/cameras/front_right_camera/image_raw/theora
/wamv/sensors/cameras/front_right_camera/image_raw/theora/parameter_descriptions
/wamv/sensors/cameras/front_right_camera/image_raw/theora/parameter_updates
/wamv/sensors/cameras/front_right_camera/parameter_descriptions
/wamv/sensors/cameras/front_right_camera/parameter_updates
/wamv/sensors/cameras/middle_right_camera/camera_info
/wamv/sensors/cameras/middle_right_camera/image_raw
/wamv/sensors/cameras/middle_right_camera/image_raw/compressed
/wamv/sensors/cameras/middle_right_camera/image_raw/compressed/parameter_descriptions
/wamv/sensors/cameras/middle_right_camera/image_raw/compressed/parameter_updates
/wamv/sensors/cameras/middle_right_camera/image_raw/compressedDepth
/wamv/sensors/cameras/middle_right_camera/image_raw/compressedDepth/parameter_descriptions
/wamv/sensors/cameras/middle_right_camera/image_raw/compressedDepth/parameter_updates
/wamv/sensors/cameras/middle_right_camera/image_raw/theora
/wamv/sensors/cameras/middle_right_camera/image_raw/theora/parameter_descriptions
/wamv/sensors/cameras/middle_right_camera/image_raw/theora/parameter_updates
/wamv/sensors/cameras/middle_right_camera/parameter_descriptions
/wamv/sensors/cameras/middle_right_camera/parameter_updates
/wamv/sensors/pingers/pinger/range_bearing
/wamv/sensors/pingers/pinger/set_pinger_position
/wamv/thrusters/lateral_thrust_angle
/wamv/thrusters/lateral_thrust_cmd
/wamv/thrusters/left_thrust_angle
/wamv/thrusters/left_thrust_cmd
/wamv/thrusters/right_thrust_angle
```

HashirZahir, 2020-05-14:
Hi can I confirm you are on the latest commit. I am not able to replicate this issue when I clone / pull the latest from this repo and rebuild the vrx simulator.  I can verify the following command publishes raw data on the relevant rostopics (including IMU data):

`roslaunch vrx_gazebo sandisland.launch camera_enabled:=true gui:=false gps_enabled:=true imu_enabled:=true`

kartikye, 2020-05-14:
Hi cloned the repository clean but I think there is something wrong with the machine im running it on. I'm still in the process of tracking down the issue.

HashirZahir, 2020-05-14:
Okay, make sure you use gazebo 9.13.0+ as older versions are not supported. This is a common source of errors

HashirZahir, 2020-05-22:
Hi is there any update on this? It would be good if you could state how you solved the problem (and close this issue) or if there are any additional issues that were faced.

caguero, 2020-05-27:
I can confirm that when launching:

`$ roslaunch vrx_gazebo station_keeping.launch vrx_sensors_enabled:=true`

I can see the lidar topic:

```
$ rostopic list | grep lidar
/wamv/sensors/lidars/lidar_wamv/points
```

and when launching:

`$ roslaunch vrx_gazebo sandisland.launch imu_enabled:=true`

I can see the IMU topic:

```
$ rostopic list | grep lidar
/wamv/sensors/imu/imu/data
/wamv/sensors/imu/imu/data/accel/parameter_descriptions
/wamv/sensors/imu/imu/data/accel/parameter_updates
/wamv/sensors/imu/imu/data/bias
/wamv/sensors/imu/imu/data/rate/parameter_descriptions
/wamv/sensors/imu/imu/data/rate/parameter_updates
/wamv/sensors/imu/imu/data/yaw/parameter_descriptions
/wamv/sensors/imu/imu/data/yaw/parameter_updates
```

I'll close the issue because we cannot reproduce it but feel free to reopen if needed.

closed 2020-05-27 by caguero