osrf/vrx #203: Navigation_task.launch

qycqycqyc, 2020-05-28:
Could you tell me how to let the usv navigate? I roslaunch navigation_task.launch,but,the usv don't move, thanks.what should I do to let the usv navigate through bouyancys.

HashirZahir, 2020-05-28:
Hi, I highly recommend looking through the [tutorials page](https://github.com/osrf/vrx/wiki/tutorials) which has a lot of information to get started. For your use case, to move the vehicle via keyboard control, simply follow this: 

https://github.com/osrf/vrx/wiki/tutorials-Driving

Hope this clarifies your question.

qycqycqyc, 2020-05-28:
> Hi, I highly recommend looking through the [tutorials page](https://github.com/osrf/vrx/wiki/tutorials) which has a lot of information to get started. For your use case, to move the vehicle via [...]

Thank you very much to answer my question，I thought when I roslaunch navigation_task.launch,the usv would navigate through buoys and avoid obstacles aotomaticlly,could you tell me how to let my thought come true. sincerely.

HashirZahir, 2020-05-28:
Hi this is a very complex question and was actually one of the challenges of the [International Virtual RobotX 2019](https://bitbucket.org/osrf/vrx/wiki/VRX%202019%20Results) where entire teams spent months developing sophisticated autonomy solutions to allow the USVs to operate autonomously to avoid obstacles. I would say this is more of a general autonomy question and is not specific to this repo. 

The general idea would be to find a way to detect the buoys (lidar + camera?) and then move to the buoys safely using some sort of global + local planner. But there are so many possible approaches to this problem as well. You can probably take a look at [some basic ROS navigation tutorials](http://wiki.ros.org/navigation/Tutorials) as a start point. Perhaps you can post/search on https://robotics.stackexchange.com/ or https://answers.ros.org/questions/ for more general autonomy questions and resources.

caguero, 2020-05-28:
Question answered, thanks @HashirZahir !

closed 2020-05-28 by caguero

bsb808, 2020-05-28:
Very well said @HashirZahir - thank you.

qycqycqyc, 2020-05-29:
Thanks for your answer.
I am a student in University,I study this major for a long time,but It is too difficult.Could you give me your automatically navigate code to me?I want to learn It very much.

Please,I can pay for it.