osrf/vrx #204: Compilation from source protobuf error

aaraujo11, 2020-06-03:
Hello,

Im tried to follow the steps to build from source, by install the [prerequisites](https://github.com/osrf/vrx/wiki/tutorials-SystemSetupInstall#install-all-prerequisites-in-your-host-system)  and compile using catkin build, but there are several compilation errors regarding ignition and protobuf. Something like:
.....
/usr/include/ignition/msgs1/ignition/msgs/world_modify.pb.h: In member function ‘const string& ignition::msgs::WorldModify::cloned_uri() const’:
/usr/include/ignition/msgs1/ignition/msgs/world_modify.pb.h:400:93: error: no matching function for call to ‘google::protobuf::internal::ArenaStringPtr::GetNoArena(const string*) const’
   return cloned_uri_.GetNoArena(&::google::protobuf::internal::GetEmptyStringAlreadyInited());
....

Any clue?

- Ubuntu 18.04.4 LTS
- ROS Melodic
- protobuf-compiler - **Version: 3.0.0-9.1ubuntu1**
-  gazebo9 -  **Version: 9.13.1-1~bionic**

caguero, 2020-06-03:
Hi @aaraujo11 , I just tested it on Docker and I got the same installed versions that you described but I was able to compile everything. Could you try to verify it on Docker maybe?

aaraujo11, 2020-06-03:
Hi @caguero, i found out the issue, basically for cartographer i need to you use the [protobuf v3.4.1](https://github.com/protocolbuffers/protobuf), i fall back to version v3.0.0 and it compile without any problem. Can you please confirm which version of protobuf are you using?

**Just in case someone have the same issue, i use this script:**

#!/bin/sh
set -o errexit
set -o verbose
VERSION="v3.0.0"
git clone https://github.com/google/protobuf.git
cd protobuf
git checkout tags/${VERSION}
mkdir build
cd build
cmake -G Ninja \
  -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
  -DCMAKE_BUILD_TYPE=Release \
  -Dprotobuf_BUILD_TESTS=OFF \
  ../cmake
ninja
sudo ninja install

caguero, 2020-06-04:
```
developer@a79dc486a777:~/vrx_ws$ dpkg -l | grep protobuf
ii  libprotobuf-dev:amd64                  3.0.0-9.1ubuntu1                    amd64        protocol buffers C++ library (development files)
ii  libprotobuf-lite10:amd64               3.0.0-9.1ubuntu1                    amd64        protocol buffers C++ library (lite version)
ii  libprotobuf10:amd64                    3.0.0-9.1ubuntu1                    amd64        protocol buffers C++ library
ii  protobuf-compiler                      3.0.0-9.1ubuntu1                    amd64        compiler for protocol buffer definition files
```

This is probably because Ignition messages was compiled with `3.0.0` and there's something going on when you try to mix versions. Glad you find the solution. I'm closing this issue then.

closed 2020-06-04 by caguero

aaraujo11, 2020-06-04:
Thank you very mcuh @caguero !!