osrf/vrx #210: roslaunch wamv_gazebo localization_example.launch...error occured

qycqycqyc, 2020-06-28:
[ WARN] [1593359201.978206712, 12.249000000]: Failed to meet update rate! Took 12.249000000000000554
[ WARN] [1593359201.978881395, 12.250000000]: Failed to meet update rate! Took 12.233333332999999143
[ INFO] [1593359201.979172024, 12.250000000]: Valid clock time received. Starting node.
[ INFO] [1593359201.983789736, 12.255000000]: Datum (latitude, longitude, altitude) is (21.309960, -157.890100, 0.000000)
[ INFO] [1593359201.983822892, 12.255000000]: Datum UTM coordinate is (615116.100362, 2356857.621317)
[ INFO] [1593359201.983854200, 12.255000000]: Initial odometry pose is Origin: (0 0 0)
Rotation (RPY): (0, -0, 0)

[ WARN] [1593359201.995698108, 12.267000000]: Could not obtain transform from wamv/imu_wamv_link to wamv/base_link. Error was "wamv/imu_wamv_link" passed to lookupTransform argument source_frame does not exist. 

[ INFO] [1593359202.004269124, 12.275000000]: Corrected for magnetic declination of 0.000000, user-specified offset of 0.000000 and meridian convergence of 0.007041. Transform heading factor is now 0.007041
[ INFO] [1593359202.004312263, 12.275000000]: Transform world frame pose is: Origin: (0 0 0)
Rotation (RPY): (0, -0, 0)

[ INFO] [1593359202.004341452, 12.275000000]: World frame->utm transform is Origin: (-631694.38578621379565 -2352468.4624987612478 0)
Rotation (RPY): (0, 0, -0.0070405899472914101195)

[ WARN] [1593359202.396014458, 12.667000000]: Transform from wamv/imu_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359202.404198909, 12.675000000]: Transform from wamv/gps_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359204.399752035, 14.667000000]: Transform from wamv/imu_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359204.425443879, 14.692000000]: Transform from wamv/gps_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359206.404334923, 16.667000000]: Transform from wamv/imu_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359206.447041018, 16.709000000]: Transform from wamv/gps_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359208.409401043, 18.667000000]: Transform from wamv/imu_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

[ WARN] [1593359208.501433232, 18.759000000]: Transform from wamv/gps_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.

HashirZahir, 2020-06-29:
Please take a look at #199 and #201 . Can you provide an image of your TF tree when you run the launch file? Also, can you show the output of `rostopic hz /wamv/robot_localization/odometry/filtered` while you run the launch file?

qycqycqyc, 2020-06-30:
[image]

qycqycqyc, 2020-06-30:
[image]

HashirZahir, 2020-07-10:
I dont see much that is wrong here. Odometry data is being updated at a good rate (60Hz). What is the problem that you are facing? The `[ WARN] [1593359208.501433232, 18.759000000]: Transform from wamv/gps_wamv_link to wamv/base_link was unavailable for the time requested. Using latest instead.` is just a warning but does not hamper the performance of the odometry calculation since those are static frames.

qycqycqyc, 2020-07-10:
@HashirZahir Emmmmm,this question does not trouble me anymore,but the odometry publisher (odom frame) is used robot localization,because of that the frame odom is not fixed on z=0.0,so,in rviz odom frame is flying, Could you tell me how to do to fix it？

HashirZahir, 2020-07-11:
I believe this behaviour can be adjusted if you follow the `robot_localization` docs where its mentioned [here](http://docs.ros.org/melodic/api/robot_localization/html/navsat_transform_node.html#zero-altitude) that setting `~zero_altitude` to true will ensure that the pose.z value is always 0. 

Hope that fixes the issue, if so, please close this issue.

qycqycqyc, 2020-07-11:
@HashirZahir  thank you so much ,when i see robot locatlization doc ,it said two_d_mode must be true,when set zero altitude,anyway,it works! thank you! it troubled me so long! GOD!

closed 2020-07-11 by qycqycqyc