osrf/vrx #211: when i roslaunch wamv_gazebo rviz_vrx.launch  the wamv/odom frame is flying

qycqycqyc, 2020-06-30:
[image]
how could i do to lock the odom frame?

qycqycqyc, 2020-06-30:
[image]

HashirZahir, 2020-07-11:
I believe this behaviour can be adjusted if you follow the `robot_localization` docs where its mentioned here that setting `~zero_altitude` to true will ensure that the `pose.z` value is always 0. Edit the parameters from here:

[localization_example.launch](https://github.com/osrf/vrx/blob/master/wamv_gazebo/launch/localization_example.launch)

Hope that fixes the issue, if so, please close this issue.

qycqycqyc, 2020-07-11:
@HashirZahir I am sorry to say it does not work.

here it is what i do in localization_example.launch

<?xml version="1.0"?>

<launch>
  
  <node pkg="robot_state_publisher" type="robot_state_publisher" name="rob_st_pub">
    </node>

  
  <node pkg="joint_state_publisher" type="joint_state_publisher" name="joint_state_publisher">
    <param name="gui" value="false" />
  </node>

  
  <node pkg="robot_localization" type="ekf_localization_node"
        name="ekf_localization" clear_params="false">
    <param name="sensor_timeout" value="2.0"/>
    <param name="two_d_mode" value="false"/>
    <param name="map_frame" value="map"/>
    <param name="odom_frame" value="odom"/>
    <param name="base_link_frame" value="base_link"/>
    <param name="world_frame" value="odom"/>
    <param name="publish_tf" value="true"/>
     <param name="~zero_altitude" value="true"/>
    <param name="frequency" value="60"/>
    <param name="imu0" value="/sensors/imu/imu/data"/>
    
    <rosparam param="imu0_config">[false, false, false,
                                   true,  true,  true,
                                   false,  false,  false,
                                   true,  true,  true,
                                   true,  true,  true]</rosparam>
    <param name="imu0_differential" value="false"/>
    <param name="imu0_remove_gravitational_acceleration" value="true"/>

    <param name="odom0" value="/robot_localization/odometry/gps"/>
    
    <rosparam param="odom0_config">[true,  true,  true,
                                   false, false, false,
                                   false, false, false,
                                   false, false, false,
                                   false, false, false]</rosparam>

    <param name="odom0_differential" value="false"/>
    <param name="smooth_lagged_data" value="true"/>
  </node>

  
  <node pkg="robot_localization" type="navsat_transform_node"
        name="navsat_transform_node" respawn="true" output="screen">
         <param name="frequency" value="60"/>
    <param name="magnetic_declination_radians" value="0"/>
    <param name="broadcast_utm_transform" value="true"/>
    <param name="~zero_altitude" value="true"/>
    <param name="wait_for_datum" value="true"/>
    <param name="use_odometry_yaw" value="true"/>
    <rosparam param="datum">[21.30996, -157.8901]</rosparam>
    <param name="yaw_offset" value="0"/>
    <param name="publish_filtered_gps" value="true"/>
    <remap to="/sensors/gps/gps/fix" from="/robot_localization/gps/fix" />
  </node>
</launch>

[image]

qycqycqyc, 2020-07-11:
<?xml version="1.0"?>

<launch>
  
  <node pkg="robot_state_publisher" type="robot_state_publisher" name="rob_st_pub">
    </node>

  
  <node pkg="joint_state_publisher" type="joint_state_publisher" name="joint_state_publisher">
    <param name="gui" value="false" />
  </node>

  
  <node pkg="robot_localization" type="ekf_localization_node"
        name="ekf_localization" clear_params="false">
    <param name="sensor_timeout" value="2.0"/>
    <param name="two_d_mode" value="false"/>
    <param name="map_frame" value="map"/>
    <param name="odom_frame" value="odom"/>
    <param name="base_link_frame" value="base_link"/>
    <param name="world_frame" value="odom"/>
    <param name="publish_tf" value="true"/>
     <param name="~zero_altitude" value="true"/>
    <param name="frequency" value="60"/>
    <param name="imu0" value="/sensors/imu/imu/data"/>
    
    <rosparam param="imu0_config">[false, false, false,
                                   true,  true,  true,
                                   false,  false,  false,
                                   true,  true,  true,
                                   true,  true,  true]</rosparam>
    <param name="imu0_differential" value="false"/>
    <param name="imu0_remove_gravitational_acceleration" value="true"/>

    <param name="odom0" value="/robot_localization/odometry/gps"/>
    
    <rosparam param="odom0_config">[true,  true,  true,
                                   false, false, false,
                                   false, false, false,
                                   false, false, false,
                                   false, false, false]</rosparam>

    <param name="odom0_differential" value="false"/>
    <param name="smooth_lagged_data" value="true"/>
  </node>

  
  <node pkg="robot_localization" type="navsat_transform_node"
        name="navsat_transform_node" respawn="true" output="screen">
         <param name="frequency" value="60"/>
    <param name="magnetic_declination_radians" value="0"/>
    <param name="broadcast_utm_transform" value="true"/>
    <param name="~zero_altitude" value="true"/>
    <param name="wait_for_datum" value="true"/>
    <param name="use_odometry_yaw" value="true"/>
    <rosparam param="datum">[21.30996, -157.8901]</rosparam>
    <param name="yaw_offset" value="0"/>
    <param name="publish_filtered_gps" value="true"/>
    <remap to="/sensors/gps/gps/fix" from="/robot_localization/gps/fix" />
  </node>
</launch>

qycqycqyc, 2020-07-11:
[image]
[image]

qycqycqyc, 2020-07-11:
[image]

qycqycqyc, 2020-07-11:
node pkg="robot_localization" type="ekf_localization_node"
        name="ekf_localization" clear_params="false"
 param name="sensor_timeout" value="2.0"/
param name="two_d_mode" value="false"/
param name="map_frame" value="map"/
param name="odom_frame" value="odom"
param name="base_link_frame" value="base_link"/
param name="world_frame" value="odom"/
param name="publish_tf" value="true"/
param name="~zero_altitude" value="true"/
param name="frequency" value="60"/
param name="imu0" value="/sensors/imu/imu/data"/

node pkg="robot_localization" type="navsat_transform_node"
        name="navsat_transform_node" respawn="true" output="screen">
         param name="frequency" value="60"/
    param name="magnetic_declination_radians" value="0"/
    param name="broadcast_utm_transform" value="true"/
    param name="~zero_altitude" value="true"/
    param name="wait_for_datum" value="true"/
    param name="use_odometry_yaw" value="true"/
    rosparam param="datum">[21.30996, -157.8901]</rosparam>
    param name="yaw_offset" value="0"/
    param name="publish_filtered_gps" value="true"/

closed 2020-07-11 by qycqycqyc