osrf/vrx #216: WAMV tf trees not publishing

808brick, 2020-09-19:
When using the VRX simulation I am unable to pull up the TF tree, and thus unable to view lidar data in RViz. I think it has to do with the fact that the joint_state_publisher node needs a `robot_description` parameter, and by the looks of it, the simulation instead sets the information to the `wamv/robot_description` parameter.

I am unable to run the [RViz Localization Tutorial](https://github.com/osrf/vrx/wiki/tutorials-Visualizing%20with%20RViz), as I just get met with errors such as:

```bash
[ERROR] [1600524585.505755739]: Could not find parameter robot_description on parameter server

[wamv/rob_st_pub-2] process has died [pid 5917, exit code 1, cmd /opt/ros/melodic/lib/robot_state_publisher/robot_state_publisher __name:=rob_st_pub __log:=/home/raymond/.ros/log/c49a7090-fa81-11ea-9f81-1002b5c27c61/wamv-rob_st_pub-2.log].
log file: /home/raymond/.ros/log/c49a7090-fa81-11ea-9f81-1002b5c27c61/wamv-rob_st_pub-2*.log

[wamv/joint_state_publisher-3] process has died [pid 5918, exit code 1, cmd /opt/ros/melodic/lib/joint_state_publisher/joint_state_publisher __name:=joint_state_publisher __log:=/home/raymond/.ros/log/c49a7090-fa81-11ea-9f81-1002b5c27c61/wamv-joint_state_publisher-3.log].
log file: /home/raymond/.ros/log/c49a7090-fa81-11ea-9f81-1002b5c27c61/wamv-joint_state_publisher-3*.log
```

If I run one of the other launch files that do not utilize the joint_state_publisher, the simulation runs, the WAMV appears and is controllable by publishing to topics, but no tf tree is detected. Running `rosrun robot_state_publisher robot_state_publisher` does not help either, as it says there is no parameter `robot_description`.

I provided a screenshot showing the tf tree in RQT and the terminal output for `rosparam list` showing no `robot_description` parameter.
[image]

808brick, 2020-09-30:
Okay I finally got everything to work. It seems the generated WAMV URDF created the GPS and IMU sensor joints as revolute joints instead of fixed. Once these were changed to "fixed" I no longer had any issues. 

I changed these two lines in my custom WAMV URDF file:
`<joint name="gps_wamv_joint" type="revolute">`
`<joint name="imu_wamv_joint" type="revolute">`

to this
`<joint name="gps_wamv_joint" type="fixed">`
`<joint name="imu_wamv_joint" type="fixed">`

This is the result:
[image]

### To Fix This Issue

1. Any localization launch files must properly configure tf_prefix parameter to `wamv`
2. Any localization launch files must properly configure robot_state_publisher so it looks at `wamv/robot_description` and `wamv/joint_states`
3. The WAMV URDF generator script `generate_wamv.launch` in the `vrx_gazebo` package needs to be modified so that GPS and IMU joint types are set to `fixed` when generated

I hope this thread is of use to the progression of the simulation.

808brick, 2020-10-13:
After closer inspection of your localization_example.launch, I realize including the `joint_state_publisher` node addresses the GPS and IMU being a revolute joint. And I attempted to get the `wamv_gazebo` `localization_example.launch` to work with a custom URDF again. The only thing that was missing the whole time was uploading the robot description.

To address this I just added the `wamv/robot_description` parameter to reference my custom urdf when I launched the simulation.  

```xml
<launch>
  <arg name="custom_urdf" value="$(find kanaloa)/config/my_wamv_5.urdf" />
  <include file="$(find vrx_gazebo)/launch/sandisland.launch" >
    <arg name="urdf" value="$(arg custom_urdf)" />
  </include>
  <param name="wamv/robot_description" textfile="$(arg custom_urdf)"/>
</launch>
```

Then I can launch the `wamv_gazebo` `localization_example.launch` node and it will work just fine. 

For others to use this, they just need to change the `custom_urdf` arg in the above launch file to the location of their custom urdf. 

To Fix The Issue:
The [Localization Tutorial ](https://github.com/osrf/vrx/wiki/tutorials-Visualizing%20with%20RViz) should be updated to include these instructions for creating a custom launch file to use a custom URDF.

mabelzhang, 2020-10-14:
Sorry no one has responded earlier. I'm not an expert with this repo, but for anyone reading this, I believe the expected way that a user would put in a custom URDF is by modifying these lines in their local copy of `vrx.launch` https://github.com/osrf/vrx/blob/7c84922e45f55a2899eca9a96652360fee512b73/vrx_gazebo/launch/vrx.launch#L51-L59

The first line sets a `urdf` parameter in the launch file to a URDF/xacro file, and the second block runs `xacro` on that file, passing in custom parameters that the xacro file takes, and throws the result on `$(arg namespace)/robot_description` on the parameter server. The `namespace` arg in the launch file is by default `wamv`, so this ends up in `wamv/robot_description`.

These lines are using a .urdf.xacro file. If that is your case, you can just replace the path in the first line. If you're using a pure URDF file, you can replace both blocks with
```
  <param name="$(arg namespace)/robot_description" textfile="$(arg custom_urdf)"/>
```
like you have already done above.

I agree the tutorial should be updated to make the experience smoother.

JTDeve, 2021-11-18:
hi, can you help me ?
my environment is ubuntu18 and melodic,code is tag 1.3.0
I also show no tf data,where is "my_wamv_4.urdf",or how i can create this urdf?

caguero, 2021-11-18:
You should get the `tf` tree published by launching:

```
roslaunch wamv_gazebo rviz_vrx.launch
```

caguero, 2021-11-19:
I'm closing this one as I think this issue is fixed but let me know otherwise.

closed 2021-11-19 by caguero