osrf/vrx #231: Low real time factor for similar beach world

maarufvazifdar, 2020-11-26:
I am working on simulating a robot in a beach environment but my real-time factor drops to around 0.5 when the robot is moving over slopes on the terrain. I am using a .dae mesh for the terrain of size 40m x40m. While running this simulation I am impressed with how stable the real-time factor remains. It will be really helpful if someone could give me some tips on improving the performance of my simulation. Thanks in advance! 

@mabelzhang @tylerlum @bsb808 @caguero

maarufvazifdar, 2020-11-26:
Update- Realtime factor drops to 0.5 with the max_step_size as 0.005 and the realtime_update_rate as 200, with the default values it dropped to 0.1 to 0.2. I am running ros-kinetic with gazebo 7.18.1

mabelzhang, 2020-11-30:
I'm not an expert in this, but it could have to do with collisions. Are you using simple collision geometries for the wheels (or whatever that is contacting the terrain)? If not, maybe try using primitives and see if there's an improvement.

maarufvazifdar, 2020-12-01:
Switching to simple collision geometries fixed the issue, earlier I was using .stl meshes for wheels and tyres, Thanks for your help. @mabelzhang

closed 2021-02-10 by maarufvazifdar