osrf/vrx #235: Dynamic Parameters

kizilkayabayram, 2021-01-13:
Hi, I would like to know dynamic system properties to design an advanced dynamic controller for the WAM-V 16 in the simulation. Some parameters that I need for this case is written in the spec sheet, such as weight and dimensions.

 However, I also checked the file which is called as ‘’wamv_gazebo.urdf’’ to do a double check. At that file, the mass value is determined as 180kg, which is different than the spec sheet (which is 204kg). Thereafter, I scrolled down to see is there any other mass value that can be added to 180kg and I saw a few. For instance, the mass values of the left engine link and left engine rear and collision are 15kg. 
Also, the same values are written for right ones. If I add all these values, it exceeds 204 kg. Hence, I want to know the real mass value that was used in this simulation. In addition, I need a few parameters more, such as inertia, center of mass, drag and friction coefficients, surface area of the vehicle that says inside of water and water density.  Could anyone point me where I can find those values?

On the other hand, I want to implement a trajectory tracking controller. For that reason, I am going to use 3 DOF dynamic model to design the controller. As far as I understood from the ‘’wamv_gazebo’’ file, 6 DOF model is used in the simulation. Do you think that it is alright to design the controller for 3 DOF and control the vehicle in the simulation?  
Thanks in advance for your response.

mabelzhang, 2021-02-10:
The mass and inertia values are in `wamv_description` https://github.com/osrf/vrx/blob/master/wamv_description/urdf/wamv_base.urdf.xacro#L124-L128

Water density and drag parameters are here https://github.com/osrf/vrx/blob/master/wamv_gazebo/urdf/dynamics/wamv_gazebo_dynamics_plugin.xacro
You can see in that file that it's using the usv_gazebo_dynamics plugin https://github.com/osrf/vrx/blob/master/usv_gazebo_plugins/src/usv_gazebo_dynamics_plugin.cc
and the ocean_waves model
https://github.com/osrf/vrx/blob/master/wave_gazebo/world_models/ocean_waves/model.xacro

kizilkayabayram, 2021-02-11:
Thanks for your comprehensive response, it really helped a lot! I have another short question about the drag parameters. Is there any way to learn the drag coefficient instead of linear and quadratic drag values? 

FYI, my controller worked in the simulation.

mabelzhang, 2021-02-11:
Glad it worked! I don't know much about how the parameters are computed. I'll leave this open in case someone else can answer it.

acxz, 2022-04-06:
Here is the computation of the drag matrix: 
https://github.com/osrf/vrx/blob/492a2a5438d513223ebff500edbcd9d6c76c0148/usv_gazebo_plugins/src/usv_gazebo_dynamics_plugin.cc#L260-L270

Currently the only way to specific the drag forces is with the linear and quad coeff terms. Anything more would involve changing the relevant drag computation code.

> learn the drag coefficient instead of linear and quadratic drag values

The drag *values* are velocity varying based on linear and quadratic *coefficients*. I believe you may have gotten these two things backwards.

M1chaelM, 2022-06-14:
@caguero This issue appears to be resolved. Can we close it before we begin the ignition migration?

M1chaelM, 2022-06-14:
@acxz No, we are still in the planning stages so there is no roadmap yet.

acxz, 2022-06-14:
Got it. I'd like to point out the following work that you may already be aware of: https://community.gazebosim.org/t/simulate-surface-waves-and-marine-vehicle-dynamics/1268

If this work could be reused and ultimately mainlined into Gazebo Sim that would be wonderful.

M1chaelM, 2022-06-21:
Closing this issue because the original questions appear to be answered. Feel free to re-open if anything remains unresolved.

closed 2022-06-21 by M1chaelM