osrf/vrx #249: Docker container starts with "/" as pwd

M1chaelM, 2021-07-07:
Following the docker tutorial [here](https://github.com/osrf/vrx/wiki/docker_build_run_tutorial) leaves me in a terminal, but `pwd` is set to root rather than the home directory:
```
Executing command: 
docker run -it   --rm    --device /dev/input/js0  --device /dev/input/js0  -v /home/username@ern.nps.edu/.gitconfig:/home/username@ern.nps.edu/.gitconfig:ro -v /home/username@ern.nps.edu:/home/username@ern.nps.edu  --name vrx_noetic_runtime   --gpus all  -e DISPLAY -e TERM   -e QT_X11_NO_MITSHM=1   -e XAUTHORITY=/tmp/.docker.xauth -v /tmp/.docker.xauth:/tmp/.docker.xauth   -v /tmp/.X11-unix:/tmp/.X11-unix   -v /etc/localtime:/etc/localtime:ro    87d9ee8be3d0 
username@78ac8e38e0cf:/$ ls
bin   dev  home  lib32  libx32  mnt  proc  ros_entrypoint.sh  sbin  sys  usr
boot  etc  lib   lib64  media   opt  root  run                srv   tmp  var
username@78ac8e38e0cf:/$ 
```
The user home information seems to be correct in `/etc/passwd` and it's easy to switch to the home directory with `cd` or `cd ~` so this is only a minor annoyance, but I'm not sure what's causing it. It may be a rocker bug. Are others encountering this error?

M1chaelM, 2021-07-07:
@bsb808 Could you check whether you see anything like this?

bsb808, 2021-08-05:
1. I think we should move the appropriate docker tutorials, i.e., those that are general to dave, vrx, etc., to the dockwater repo rather than have them spread across and repeated in the various project wikis.
2. Yes, I see the same thing.  I am just in the habit of issuing `cd` in each new terminal.  Perhaps a feature request for rocker?

M1chaelM, 2021-08-05:
@bsb808 Thanks for taking a look at this. In reverse order: Yes, now that I know this is not a local problem, I think it is a rocker bug. I don't think the intended behavior is to start a user with the working directory at the root of repository. This sounds like something is failing somewhere.

Regarding the Docker tutorials, I have already created a generic tutorial on the dockwater website and edited the various wikis to link to it when possible. This makes the most sense when explaining how to set up the system to run docker and rocker, etc., described here: https://github.com/Field-Robotics-Lab/dockwater/wiki/Install-Dependencies

However, the instructions for building and running the image in the dockwater repository are written for the general case and contain details explaining how a user might want to customize them depending on what they are doing. For dave and vrx we already know what they are doing so it is pretty easy to give a quick version specific to the use case, so that's what I've done. This creates a little repetition in exchange for a gain in efficiency, which I think is a worthwhile tradeoff. It also makes it easy to integrate the instructions into a continuous flow so the user doesn't have to use the back button on the browser to get back to where they are doing.

It is possible that there is some more streamlined way of doing things but I think I need to use it a bit before I have a sense of what this is. In the meantime, since the instructions work I think we should focus on the release tasks.

M1chaelM, 2021-08-10:
@j-herman and @crvogt I'm no longer able to reproduce this issue. Could you run through the tutorial below and verify that you end up in a command prompt with your working directory set to home (~) rather than (/)?
https://github.com/osrf/vrx/wiki/docker_build_run_tutorial

I'm not sure what changed since Brian checked last week, but if no one can reproduce this I will close it.

M1chaelM, 2021-09-09:
@bsb808 Can we close this?

M1chaelM, 2021-09-24:
@bsb808 reports he can no longer reproduce. @caguero if this is no longer happening after you rebuild your image I think we can close it.

caguero, 2021-09-24:
> @bsb808 reports he can no longer reproduce. @caguero if this is no longer happening after you rebuild your image I think we can close it.

I just rebuilt my image and I don't have the issue anymore :) .

closed 2021-09-24 by caguero