osrf/vrx #257: WAMV missing from SandIsland Launch

M1chaelM, 2021-07-09:
One of our recent pull requests has broken `sandisland.launch`. Attempting to launch it now loads the Sydney arena instead of Sand Island, and the WAM-V is missing:

[image]

Although Sand Island is no longer strictly needed for VRX 2022, it would be nice to still be able to use it for basic demos and tutorials involving the VRX platform.

M1chaelM, 2021-07-09:
@caguero Could you take a look at this?

crvogt, 2021-07-09:
@M1chaelM It looks more like a GPS issue, the WAM-V is there, just not at the camera location at the start. I can set it up with the WAM-V in the correct location, unless you'd rather revert to the sandisland map.

M1chaelM, 2021-07-12:
@crvogt Thanks--yes, I think changing the environment that the `sandisland.launch` file pulls up is probably an unintended consequence of recent changes (and maybe a problem with the way the launch file was originally written). It should continue to bring up Sand Island (Honolulu).

caguero, 2021-07-12:
This issue should be fixed in https://github.com/osrf/vrx/pull/256. I forgot to update a launch file.

M1chaelM, 2021-07-14:
Confirming this is now fixed. Thanks!

closed 2021-07-14 by M1chaelM