osrf/vrx #258: Robot Localization Example Tweaks

M1chaelM, 2021-07-09:
Our robot localization example now works as described in PR #247 but as the screenshot shows, Rviz is placing the WAM-V very far away from the reference grid:

[image]

Can we center the grid?

Also, running
```
roslaunch wamv_gazebo localization_example.launch
```
still produces a disconcerting stream of warnings. I realize this is a topic of hot discussion, but until it's resolved can we do something to suppress them? I don't want our users to think they've done something wrong if this the expected behavior. If suppressing the warnings causes problems let's at least add some documentation to the tutorial.

M1chaelM, 2021-07-09:
@crvogt Tagging you since this is relevant to the tutorial you're reviewing.

crvogt, 2021-07-09:
@M1chaelM I'll look at re-centering the grid. With regards to the warnings, I can add a line to the launch that should suppress the warnings.

M1chaelM, 2021-07-12:
@crvogt Great! Thank you.

crvogt, 2021-07-15:
@M1chaelM I'm noticing that running `rostopic echo /wamv/robot_localization/odometry/filtered` in the rviz example that the `pose` is not only very large, but suggests the wamv is moving. Have you come across this/is this expected?

crvogt, 2021-07-16:
@M1chaelM After restarting my machine, the pose is fine. I'll do a PR with the grid updated to match the location of the wamv.

M1chaelM, 2021-07-16:
A slightly more targeted error (but still very clumsy) suppression line is:

```
2> >(grep -v TF_REPEATED_DATA)
```
I think we can add it to the command line when needed, rather than baking it in to anything. Since the grid location is fixed with PR #267 I will close this issue.

closed 2021-07-16 by M1chaelM