osrf/vrx #261: Update to Python files to use Python3

M1chaelM, 2021-07-09:
## Context
* Since we are moving to Ubuntu 20.04, python 2 is no longer installed by default.
* We could install it, but this adds dependencies and some amount of confusion keeping the versions straight.
* We only have about a dozen python files and it is not too hard to move them to python 3 (in many cases this is just a matter of changing outdated executable paths at the top of the script).

## Proposed
Let's make our code work with Python 3.

crvogt, 2021-07-09:
@M1chaelM This seems like something I can take care of.

M1chaelM, 2021-07-14:
It looks like our acoustic pinger tutorial may rely on this changes, since the `pinger.launch` file fails with the Python error:
```
ImportError: No module named yaml
```
This comes from both the `pinger_visualisation.py` and `set_pinger_position.py` nodes. The yaml module is definitely installed so I think these are trying to run python 2.

crvogt, 2021-07-15:
@M1chaelM I found that by simply modifying the shebangs at the top of the python files to python3 the pinger tutorial work fine. I haven't tested in Melodic though. I think ideally there'd be a parameter to use one or the other, but for a quick fix, this does work. Thoughts?

M1chaelM, 2021-07-15:
I think changing to force running in Python 3 should be fine. This will just mean that people running in Melodic need to install Python 3, which shouldn't be too much of a burden. It looks like @caguero did find a couple places where the code needs to be updated to eliminate things that aren't supported in Python 3, so it may be worth skimming through to look for these in case they don't pop up right away in testing.

crvogt, 2021-07-15:
@M1chaelM Sounds good!

M1chaelM, 2021-07-16:
Fixed in PR #265 and #266.

closed 2021-07-16 by M1chaelM