osrf/vrx #271: twist2thrust.py:

kledom, 2021-07-19:
I'm using VRX to test ASV control algorithms and have some questions regarding the twist2thrust.py node.

After starting the node with `roslaunch vrx_gazebo usv_joydrive.launch` I realized that the robot is permantly steering to the right upon receiving linear.x commands. I looked into the code and found the following snippet:

https://github.com/osrf/vrx/blob/a8037d3120795ef5b39e571718a8d5e0c84c8fee/vrx_gazebo/nodes/twist2thrust.py#L33-L40

So, when the keyboard parameter is `false` - which is the case in usv_joydrive.launch - linear.x is only send to the left thruster. The right thruster is responsible for handling angular.z. Is this the intended behavior, since it means that you have to permantly send angular.z=1 in order to drive straight ahead

https://github.com/osrf/vrx/blob/a8037d3120795ef5b39e571718a8d5e0c84c8fee/vrx_gazebo/nodes/twist2thrust.py#L29-L31

bsb808, 2021-07-19:
Thank you for the issue.  We certainly welcome pull requests.

If you are using the off-the-shelf setup for VRX, then it would be ideal of the PR could be tested using one of our tutorials, e.g., https://github.com/osrf/vrx/wiki/driving_teleop_tutorial

My suspicion is that the snippet you highlight is an artifact of the interplay between the joy node mapping and the mapping to individual thruster commands.  

All the best.

bsb808, 2021-08-04:
I verified that the teloperation tutorials work as expected: https://github.com/osrf/vrx/wiki/driving_teleop_tutorial
I also added some detail to the wiki.

bsb808, 2021-08-04:
Closing this b/c it appears operational as-is.  @kledom - please re-open if there is something not working on your end.  Again, always happy to receive PRs.

closed 2021-08-04 by bsb808