osrf/vrx #277: Update task 5 - Dock

caguero, 2021-07-22:
* Load Sydney Regatta instead of Sand Island
* Update the starting pose of the WAM-V
* Update the task to use the RobotX 2022 dock

M1chaelM, 2021-08-03:
Copy from vorc:
* `gymkhana.launch`
* `gymkhana.world.xacro`

Update:
* `gymkhana.launch`
* `gymkhana.world.xacro`
* `pinger.yaml`

M1chaelM, 2021-08-03:
Getting warnings about non-unique names in the models:
```
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[short_navigation_course_0].
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[buoys].
```

M1chaelM, 2021-08-03:
Notes about updating `short_navigation_course0`:
* We no longer have a blue marker, so the exit gate is black on the port side instead.
* Instead of port and starboard, the model names the gates "red" on the port side and "green" on the right. Is it a nautical convention to call the port gates "green"? If not, this is a little confusing because not all of our green gates are green. The names of these gates are referenced external to the model when they are passed in to the navigation scoring plugin so it might be worth renaming.

M1chaelM, 2021-08-03:
Notes about updating `obstacle_course` to use new course elements:
* Previously this used 3 different sizes of round buoys:
  * `polyform_a3`: smallest
  * `polyform_a5`: medium size
  * `polyform_a7`: largest
* The new course elements have only two types of round buoys and they are both the same size:
  * `mb_round_buoy_black`
  * `mb_round_buoy_orange`

I've updated to replace all the `polyform_a3` and `polyform_a5` with `mb_round_buoy_black` and `polyform_a7` with `mb_round_buoy_orange`.

M1chaelM, 2021-08-03:
Problems encountered while running through tutorial:
* The high level `task/info` topic does not appear in the list.

M1chaelM, 2021-08-03:
The `wamv::base_link` errors are resolved by changing the lock joint in `wamv_gazebo.urdf.xacro` to `wamv::wamv/base_link`. It looks like the prefix `wamv/` was added to all links except `wamv_external_link`, but a few places were missed.

A similar change to the wind object `link_name` in the `gymkhana.world.xacro` makes the debug message complaint disappear. This is probably also why we weren't seeing any wind in the default tutorial.

M1chaelM, 2021-08-04:
TODO:
* Polish tutorial language.
* Create PR

M1chaelM, 2021-08-10:
Resolved with PR #307

closed 2021-08-10 by M1chaelM