osrf/vrx #279: Include RobotX 2022 elements when launching vrx.launch

caguero, 2021-07-22:
In the past, when launching `sandisland.launch` the simulation loaded an empty Sand Island world with a very basic WAM-V without sensors. The old `vrx.launch` loaded the WAM-V in the competition configuration (with all the sensors) and a Sand Island world with multiple RobotX elements to practice.

Right now, launching `sydneyregatta.launch` has a consistent behavior as it loads a basic WAM-V and an empty Sydney Regatta world. However, launching `vrx.launch` loads the exact same world as `sydneyregatta.launch` without any RobotX elements. We need to update the code to include some of the updated elements (docks, navigation channel, etc.).

M1chaelM, 2021-09-09:
@bsb808 Could you take a look at this?

bsb808, 2021-09-10:
@M1chaelM I will try, but I don't know if I'll have time by COB tomorrow.  I spent most of the day on issues and PRs a related to the video demos and have a pretty hectic day tomorrow.   I'll try to take a look later tonight.

M1chaelM, 2021-09-10:
@bsb808 OK, thanks! If you can't get to it let me know and we'll figure something out.

M1chaelM, 2021-09-10:
@bsb808 I can work on this unless you've already started on it.

bsb808, 2021-09-11:
@M1chaelM Cool.  I haven't started.   This will likely need to be re-worked with PR #358

closed 2021-09-12 by caguero