osrf/vrx #284: Update competition documents

caguero, 2021-07-22:
Create new versions of the `VRX Competition and Task Descriptions` and `VRX Technical Guide` according to the VRX 2022 specifications.

M1chaelM, 2021-07-29:
I will take an initial pass through the documentation and create sub-tasks if needed.

M1chaelM, 2021-08-02:
Started draft of VRX 2022 Technical Guide here (Google drive access required): https://docs.google.com/document/d/1szkmWh2HpOC4AR0wFJdjP4t5FT4jpKqQ/edit#heading=h.35nkun2
* Most of this document requires only minor revision.
* Biggest exception is section 8, which describes the API. This will likely change to support new tasks.

M1chaelM, 2021-08-02:
Started draft of VRX 2022 Task Descriptions here (Google drive access required): https://drive.google.com/file/d/1a2-L0UeZKFq4RX283a22PO2gGTAXVIqV/view?usp=sharing
* Sections 1-3 require minor revisions.
* Section 4 requires major revision, especially section 4-6

M1chaelM, 2021-08-09:
First pass update of VRX 2022 Task Descriptions, sections 1--4.1.2

M1chaelM, 2021-08-25:
Version 1.0 of the VRX 2022 Technical Guide is now ready.