osrf/vrx #286: Create a tutorial for validating the submission

caguero, 2021-07-22:
Document the steps to double-check that a submission is correct. Use our trivial solution as an example.

M1chaelM, 2021-09-10:
This tutorial is now complete: https://github.com/osrf/vrx/wiki/vrx_2022-validation

It can be tested using the three submission files in the example_team submission [uploaded here](https://github.com/npslearninglab/vrx-events/tree/2022_rehearsal_team_osrf/2022/phase2_dress_rehearsal/example_team)

@j-herman, do you have time to run through this and make sure it works for you? I'd like at least one person to double check it before I close the issue.

j-herman, 2021-09-10:
I checked everything out - made one minor change to the tutorial to clarify that files need to be in a folder named <your_team_name>.  All worked as advertised.

M1chaelM, 2021-09-10:
> I checked everything out - made one minor change to the tutorial to clarify that files need to be in a folder named <your_team_name>. All worked as advertised.

Great! Thank you for catching that. I added a step to give the command to create the directory.

@caguero Since @j-herman was able to review this I will close it now.

closed 2021-09-10 by M1chaelM