osrf/vrx #287: Create a tutorial for running the competition and evaluating a submission

caguero, 2021-07-22:
Running the competition evaluating all the submissions is a task that we do to run the competition. Before that, it's important to document the process and show the teams how to run it by themselves. They'll be able to catch potential issues with their submissions and get real scores.

M1chaelM, 2021-09-10:
* A new tutorial is now available here: https://github.com/osrf/vrx/wiki/vrx_2022-testing
* It can be tested simultaneously with https://github.com/osrf/vrx-docker/pull/34

Notes:
* Our world-generation scripts are currently broken due to issue #348, so I've reworked the tutorial to use pre-built worlds.
* We have not yet made the VRX 2022 worlds available (these were planned for 9/24) so the tutorial uses the VRX 2019 practice worlds.

M1chaelM, 2021-09-21:
I'm not closing this yet because we should update the tutorial to use 2022 practice worlds when #331 is merged.

M1chaelM, 2021-10-05:
Now updated to use 2022 practice worlds.

closed 2021-10-05 by M1chaelM