osrf/vrx #289: Update the automatic evaluation tool

caguero, 2021-07-22:
Our set of scripts will need to be updated to account for the new tasks and changes in the scoring rules.

M1chaelM, 2021-09-10:
Status:
* I have updated the base tool image to run on noetic and this now appears to be working.
* Due to our reliance on a bug in xacro which has now been fixed, our generated world.xacro files don't work.
* We don't yet have .yaml files to generate these worlds from anyway.
* We are in the process of creating practice worlds "by hand" and these should work.

In light of the above, I propose to merge changes to update the automatic evaluation tool to noetic and push the automatic world generation part to a future release.

M1chaelM, 2021-09-10:
The update to the base image can now be tested at https://github.com/osrf/vrx-docker/pull/34

When this is merged I'll move the rest of this issue to a future release.