osrf/vrx #300: Documentation for Task 4: Wildlife Encounter

M1chaelM, 2021-08-03:
The Wildlife Encounter task is new and needs to be documented in the VRX 2022 Task Descriptions. The documentation should include description of:
* Capability being tested
* Implementation
* Scoring

M1chaelM, 2021-08-04:
Sketch of the general plan for this task:
* launch an initial world with 3 "animals" somewhere nearby
* Plugin controls animal movement
  * Give it a few waypoints and a fixed velocity and have the animals move through them on a loop.
  * Simplest version is a single waypoint so the animal doesn't move
* Use a logical camera sensor to allow system to query a field of view and get back a list of animals in the view
* Scoring plugin must detect:
  * Successful clockwise loop around a platypus.
  * Successful counterclockwise loop around a turtle
  * Successful avoidance of crocodile.

Tentative Scoring Plugin Implementation Strategy:
* Detect when the WAM-V has completed any loop. (i.e. crosses its own path)
  * As the vehicle moves, sample regularly from its location to create a set of points.
  * Treat this set of points as "gates"
  * Detect when the vehicle crosses through a gate and in what direction  
* Determine the direction of the loop (clockwise / counterclockwise)
  * Of the points that make up a notional gate, one will have a more recent timestamp than the other.
  * If the vehicle passes through the gate with the more recent timestamp on the port side, that is a clockwise loop.
  * If the more recent timestamp is on the starboard side, that is a counterclockwise loop.
* Determine whether the appropriate animal was "inside" the loop.
  * Once a loop is detected, calculate the region described by the points.
  * Check the location of the animal to determine whether it is in the region.
  * Note: this method means it's possible for the animal to escape the loop if you take too long to close it.
  * If that is too unforgiving we could also look at some prior positions of the animal.
* For the crocodile, identify a region around it and detect whether the vehicle is in the region.

Tentative Scoring suggestion:
* The task is complete when it either times out, or when both the turtle and the platypus have been correctly circled.
* The score is equal to the task completion time.
* There is a time penalty added for getting too close to the crocodile.
* There is a time deduction for each correct circle. 
* Lowest score is best.

M1chaelM, 2021-08-04:
@caguero @bsb808 Thoughts on the sketch above? It is more detailed than needed for the documentation but I wanted to get it worked out in my head first.

M1chaelM, 2021-08-04:
I just thought of the following alternative Strategy for detecting scoring events, which may be superior and easier to implement:
* For each animal, create a ring of "spokes" around it made up of rectangular regions.
* Detect when the vehicle passes through a spoke, and in what direction.
* If the vehicle crosses through a spoke that's associated with the crocodile, that counts as a deduction.
* For platypus and turtle, we would need to detect that all gates are crossed in the correct direction and order, and probably within some limited amount of time.

Advantages of this approach:
* It is easier to make sure the WAM-V is actually close to the wildlife it is supposed to be circling, which seems in the spirit of the task.
  * Using the other implementation, we could try to achieve this by restricting the side of the region but the math could be a little more complicated.
* It is probably easier to create a visualization for debugging / practice. 

Disadvantages:
* Not as many "loop" shapes will work.

j-herman, 2021-08-05:
I had to draw option 2 out to see how it would work... I  like it.  You can just run a line segment-intersection on pairs of sequential points in the vehicle path, which seems a lot easier than keeping track of all potential gates.  Might need to play with the update frequency to make sure you don't miss a spoke when the two are moving in opposite directions.
[image]

M1chaelM, 2021-08-05:
I thought it might be useful to explain how the implementation is relevant to task documentation, since that's the actual issue this is supposed to address.

Having thought about the options above, I propose a successful circling event should have the following properties:
* The animal is in the interior of the circle.
* The vehicle must remain within some fixed distance of the animal while circling (i.e. you can't just circle the whole area and be 100 meters away).
* The circle has to be completed in one go--you can't drive part way around the animal and then come back and do the other side later. This is what I was trying to capture with the "timeout" suggestion but it might be simpler just to say that if you get too far away you have to start over.
* If you start circling in one direction and then change your mind and go the other direction, we should treat this as starting the circle over.

Things I'm not yet sure about:
* How close to the animal is close enough? 2 meter radius?
* Likewise, how close is too close to the crocodile? (same as above?)
* What should the time penalties and bonuses be?

I'd love feedback on this from people who have a better sense of what's reasonable for a real WAM-V. In the meantime I will draft the documentation without specifying these parameters yet.

M1chaelM, 2021-08-06:
Almost forgot: I assume actual collision with the animals is not desirable. Include a penalty for this?

j-herman, 2021-08-09:
Looking through the documentation on the real WAM-V, the minimum turn radius appears to be ~2.5 meters.  Given that the animal may be moving and that we've got some variability in sea state, some additional buffer would be good.  Might we want to adjust both the minimum and maximum distances based on environmental conditions?  
Thinking about how to set this up and score it, I'm wondering what specific capabilities we're most interested in evaluating.  Are we more concerned about precision maneuvering, efficient task completion, target tracking, or ...?

M1chaelM, 2021-08-09:
@j-herman Thanks for looking at that. Do you have a link? I'm very open to a larger radius but also confused about what is meant by turning radius in this case since the thrusters can go in opposite directions at the same time.

The primary goal is to combine basic capabilities from tasks 1-3 (station-keeping, path planning, and percpetion) into an application that mimics a real world requirement (observe / don't get eaten by animals). A secondary goal is to pave the way for additional capabilities that will be tested in RobotX 2022.

Typically, we want to pick a goal that is relatively easy to accomplish via teleop in easy environmental conditions (flat sea, stationary animal, no wind), then just expect it to get harder as those conditions become more challenging. We can adjust the goal if needed during testing, we just have to do this before we publish the requirements.

j-herman, 2021-08-09:
Oh, that makes more sense!  I misinterpreted "within its own length" and the only actual numbers I found must have been for a different propulsion setup.  Ignore that comment.  I do still think you'll want more clearance than 2 meters if you're looking at the center point of the WAM-V for distance measurements, but if you measure off of a bounding box it could be a good place to start.
Probably best just to build the task and then evaluate, as you suggested.