osrf/vrx #302: Create Wildlife "Buoys"

M1chaelM, 2021-08-03:
We need to request design of 3 additional objects that will represent the wildlife in our wildlife encounter. They don't need to be realistic (cartoonish is fine), but should be visually identifiable as the following 3 animals:
* Crocodile
* Platypus
* Turtle

caguero, 2021-08-05:
I forgot how this buoy should be. Should this be a buoy with a placard containing the animal or a buoy with the shape of an animal?

M1chaelM, 2021-08-09:
It has not been decided. Since we will use the logical camera for identification, I think a buoy with the shape of an animal might be preferable.

caguero, 2021-08-17:
Here are some concepts: 
[image]