osrf/vrx #303: Update cpplint.py to match Gazebo 11's version

M1chaelM, 2021-08-03:
Our `cpplint.py` file seems to have diverged from other versions used at OSRF. See [discussion at PR 140 of the Dave repository](https://github.com/Field-Robotics-Lab/dave/pull/140).

At the moment I think we have lint checks disabled. This is OK but could eventually lead to a build-up of style issues.

## Proposed Solution:
Take the same approach used in the Dave PR above and copy the cpplint.py from Gazebo 11.

M1chaelM, 2021-08-03:
@caguero What do you think about the fix suggested?

caguero, 2021-08-05:
Is the Gazebo 11 `cpplint.py` compatible with Python3?

M1chaelM, 2021-08-09:
Running `pylint --py3k` suggests that the Gazebo 11 version is a little more Python3 compatible than what we are using now. There are 8 issues that should be pretty easy to fix to get the same functionality. I could do this as part of the update.

M1chaelM, 2021-10-21:
Resolved with #374

closed 2021-10-21 by M1chaelM