osrf/vrx #312: GPS coordinate mismatch

crvogt, 2021-08-12:
The position of the WAM-V in simulation doesn't match map lat/lon.

Modified `wamv_gps.xacro` with world origin lat/lon

Changing the `<referenceHeading>` field in `wamv_gazebo.xacro` seems to have a corrective effect in terms of direction, but the length scale is wrong.

Changing parameters in the `sydneyregatta.xacro` `<spherical_coordinates>` field hasn't had an effect for me.

WAM-V starting position via Google Maps
[image]
World Origin via Google Maps
[image]
WAM-V starting position in Gazebo
[image]
WAM-V at origin in Gazebo
[image]

crvogt, 2021-08-12:
Setting the `<referenceHeading>` field in `wamv_gps.xacro` to 270 puts the WAM-V at the correct position. Will now check that placing buoys workds in the world via lat/lon

closed 2021-08-16 by crvogt

M1chaelM, 2021-09-10:
This was resolved with #313