osrf/vrx #318: Installing on Melodic fails

AntonBock, 2021-08-18:
Hello, I am trying to install the package on a Melodic system. I have followed your tutorial, with the exception of changing "gazebo11" to "gazebo9" and "noetic" to "melodic" in all the command lines. 

When making the packing, i get the following error:

```c[ 26%] Linking CXX shared library /home/ros/vrx_ws/devel/lib/libHydrodynamics.so \nl
[ 26%] Built target Hydrodynamics
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

I am assuming it's the compatibility with Melodic which is the issue. Is there a somewhat easy fix, or do I need to upgrade my entire system? 

Br Anton

M1chaelM, 2021-08-18:
We officially only support Noetic. However, it is possible to install on Melodic, and there is a working example using the Docker-based install path, and we test that the code builds on both Noetic and Melodic. The dependencies are slightly different, though.

A few things that would help:
* I think you are saying you've followed this tutorial, using Option A: https://github.com/osrf/vrx/wiki/system_setup_tutorials Is that right?
* Could you post a longer snippet of your output so we can see the cause of the failure?
* Can you try the Docker-based install? This would allow you to run the Noetic environment without upgrading your host OS.

AntonBock, 2021-08-19:
Thanks for taking the time to reply. When looking through the error message again this morning, I found that the build failed earlier, because of an outdated version of numpy. Installing it with `python3 -m pip install numpy` fixed the error in my issue. 

Thanks again, and sorry!

closed 2021-08-19 by AntonBock