osrf/vrx #327: Why the process has died

boeun98, 2021-08-30:
I want to revise the wave variables.

So I revised 'vrx/wave_gazebo/world_models/ocean_waves/model.xacro'. 

However, if I modify the variables (such as, gain, period, direction_x, direction_y, angle, scale), the process is dead.

Why process has died?

-------------------------------------------------------------------------------
(vrx/wamv_gazebo/urdf/dynamics/wamv_gazebo_dynamics_plugin.xacro)

And, if I modify the variables (such as, width, length, xDotU) a little about the hydrodynamics, the position of the robot model is changed to (0, 0, 0).

Also, if I check the rqt_graph, the connection is broken.

What's wrong with this?

Thank you.

caguero, 2022-01-25:
Hi @boeun98 . I went ahead and modify `wave_gazebo/world_models/ocean_waves/model.xacro`. I changed the gain from `0.1` to `1`. Then, recompile and run it with `roslaunch vrx_gazebo vrx.launch` and Gazebo seems to run properly. I'm not sure what's your issue but I can't reproduce it.

caguero, 2022-02-22:
Closing this issue due to lack of activity. Please reopen if needed proving extra information.

closed 2022-02-22 by caguero