osrf/vrx #334: Built target Hydrodynamics make: *** [Makefile:141: all] Error 2

siddarth09, 2021-09-08:
I have no idea why I am getting this error, can someone help me out

M1chaelM, 2021-09-10:
Could you check whether the issue you're experiencing is captured (and resolved) by 
* #355?

siddarth09, 2021-09-12:
[ 24%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/WavefieldEntity.cc.o
[ 24%] Linking CXX shared library /home/sid/vrx_ws/devel/lib/libHydrodynamics.so
[ 24%] Built target Hydrodynamics
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

siddarth09, 2021-09-12:
I am still getting the same error even after I create a new workspace

M1chaelM, 2021-09-14:
I need to know a bit more to help troubleshoot:
* Were you previously able to build the workspace?
* Are you running on Ubuntu 20.04?
* Also, are you using the host-based install or Docker?

siddarth09, 2021-09-15:
I am using ros melodic on ubuntu 18 (virtual box-virtual machine), this is the first time I built my workspace, following the tutorials

M1chaelM, 2021-09-16:
OK, please see the system requirements: https://github.com/osrf/vrx/wiki/system_requirements. We recommend that you use Ubuntu 20.04, and that is what the tutorials are written to support. It is possible to run on Ubuntu 18 but the dependencies are different.

siddarth09, 2021-09-17:
Yes i checked now with my noetic distro and it is working, thanks a lot

closed 2021-09-17 by siddarth09