osrf/vrx #347: Scan and dock - Missing and extra ROS topics

bsb808, 2021-09-09:
Following this example - https://github.com/osrf/vrx/wiki/vrx_2022-scan_dock_deliver_task

Two of the three ROS topics mentioned do not seem to exist.  Missing
```
/vrx/task/info
/vrx/scan_dock_deliver/color_sequence
```

As I recall, the second topic is how we change the light buoy color seq.   Need to have this feature to be able to generate the highlight video.

M1chaelM, 2021-09-09:
@bsb808 Can you check that you have the most recent version of the code? The PR for this task was just merged 2 hours ago. When I run it I can see the `/vrx/task/info` topic and the `/vrx/scan_dock_deliver/color_sequence` service. Also, this highlight video was assigned to me and I have already generated it.

M1chaelM, 2021-09-09:
@bsb808 Edit: I found your note in the spreadsheet explaining you want to use this functionality for the video snippet showing the new docks, so disregard that last comment. Let me know if this still doesn't work for you after pulling and rebuilding.

bsb808, 2021-09-10:
@M1chaelM Awesome .  Yes, the example now works much better!

I think the wiki needs to be modified (I'm not positive - if I was, I'd go ahead and make the change)

1. Step 2: Subscribe to relevant topics:  I believe this should be more general since we are dealing with both messages and services with the API.
2. `/vrx/scan_dock_deliver/color_sequence`  I see the ROS service, but not a ROS topic publication.  Is this expected?  If so, I think we should separate the messages and services in the explanation.
3. For shooting a ball, publish a message - Suggest being more specific with "For shooting a ball, publish a [std_msgs/Empty](http://docs.ros.org/en/melodic/api/std_msgs/html/msg/Empty.html) message".  Perhaps this is too much information and redundant with the competition documents?
3

M1chaelM, 2021-09-10:
@bsb808 Thanks! Tutorial updated. Closing this issue.

closed 2021-09-10 by M1chaelM