osrf/vrx #350: Triplicate state messages in gymkhana task

M1chaelM, 2021-09-09:
Because we are using a scoring plugin that combines two other scoring plugins, the gymkhana task prints state output in triplicate:
```
[Msg] OnReady
[Msg] Vehicle released
[Msg] OnReady
[Msg] Vehicle released
[Msg] OnReady
[Msg] Publishing Goal coordinates
[Msg] OnRunning
[Msg] OnRunning
[Msg] OnRunning
```
This may not cause a problem with functionality but at best it is a little confusing.

M1chaelM, 2021-09-10:
Related to this, completing the navigation portion of the course produces the following error:
```
[Err] [scoring_plugin.cc:470] VRX_EXIT_ON_COMPLETION and <per_plugin_exit_on_completion> both not set, will not shutdown on ScoringPlugin::Exit()
[ERROR] [1631232984.088772410, 43.565000000]: VRX_EXIT_ON_COMPLETION and <per_plugin_exit_on_completion> both not set, will not shutdown on ScoringPlugin::Exit()
```
I assume this is expected but it is also a bit confusing.

caguero, 2021-09-10:
I modified these messages to contain the name of the plugin. See a1e5f1e. However, when a plugin doesn't override the base `ScoringPlugin` class, the message that we'll see is going to be a bit more generic. Here's how it looks now:

```
[Msg] ScoringPlugin::OnReady
[Msg] ScoringPlugin::Vehicle released
[Msg] ScoringPlugin::OnReady
[Msg] ScoringPlugin::Vehicle released
[Msg] StationkeepingScoringPlugin::OnReady
[Msg] <StationkeepingScoringPlugin> Publishing Goal coordinates
[Msg] ScoringPlugin::OnRunning
[Msg] ScoringPlugin::OnRunning
[Msg] StationkeepingScoringPlugin::OnRunning
```

About the error message after completing the navigation portion: This is expected because this plugin thinks that we forgot to set a termination condition. In this particular task (I think it's the only one), we rely on the upper level plugin (gymkhana_score_plugin) to shutdown Gazebo.