osrf/vrx #353: Is the Sydney Regatta map orientation correct?

bsb808, 2021-09-10:
I may be confused, but I think the Sydney Regatta map is incorrectly oriented.  

I launch the regatta world with the USV at the origin...

```
roslaunch vrx_gazebo sydneyregatta.launch x:=0 y:=0 Y:=0
```

Here is what I believe is the birds eye view of the location with North up.
[image]

And here is the google map view...
[image]

Are we 180 degrees off?  

Is this as simple as applying a Yaw rotation to the model origin?

M1chaelM, 2021-09-10:
Uh oh, I wonder if this is related to:
* #312 

@crvogt noticed last month that the GPS coordinates were reversed. The PR is not referenced in the issue, but I will look for it to see whether this is a side effect of our solution.

M1chaelM, 2021-09-10:
Update: #312 was resolved with:
* #313
Which rotated the arena 180 degrees in this commit: https://github.com/osrf/vrx/pull/313/commits/b574656fe9a2e4f6ab6e3f99485e83a6f095c950

I could be misunderstanding, but it seems like this fix had the unintended consequence you are noticing here.

bsb808, 2021-09-10:
> Update: #312 was resolved with:
> 
> * [Update Station Keeping Task #313](https://github.com/osrf/vrx/pull/313)
>   Which rotated the arena 180 degrees in this commit: [b574656](https://github.com/osrf/vrx/commit/b574656fe9a2e4f6ab6e3f99485e83a6f095c950)
> 
> I could be misunderstanding, but it seems like this fix had the unintended consequence you are noticing here.

The first thing to make sure we have correct is the Gazebo coordinates (ENU) and how they relate to the Sydney map.  The spherical_coordinates parameters should be independent of this .  I don't believe that changing the <heading_deg> parameter will affect the gazebo coordinate orientation.

caguero, 2021-09-10:
I see a 180 deg. rotation applied in the world [here](https://github.com/osrf/vrx/blob/master/vrx_gazebo/worlds/sydneyregatta.xacro#L41).

caguero, 2021-09-10:
I think you're right about this. I believe we need to undo the yaw rotation for the SydneyRegatta model in the world. [Here](https://github.com/osrf/vrx/blob/master/vrx_gazebo/worlds/sydneyregatta.xacro#L41).

Then, assuming that our GPS coordinates are working correctly right now, I guess, we'll need to undo the `<heading_deg>180</heading_deg>` in the `<spherical_coordinates>` section, right? [Here](https://github.com/osrf/vrx/blob/master/vrx_gazebo/worlds/sydneyregatta.xacro#L22).

bsb808, 2021-09-10:
I'm hoping that the two things - the geodetic-mapping and the orientation of the model - are independent.  Trying now.

closed 2021-09-21 by M1chaelM