osrf/vrx #355: libHydronamics fails to build

M1chaelM, 2021-09-10:
I can no longer build the master branch of the VRX repository. The build fails with the following error:

```
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libsdformat9.so.9.6.0', needed by '/home/mrmccarr@ern.nps.edu/vrx_ws/devel/lib/libHydrodynamics.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:1509: vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
Anyone know what happened? This is very recent so I'm hoping it's an easy fix.

M1chaelM, 2021-09-10:
I think I missed some important error messages in the original post. Here's the full output:
```
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] xacro: generating worlds/buoyancy_plugin_demo.world from worlds/buoyancy_plugin_demo.world.xacro
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] xacro: generating urdf/wamv_base.urdf from urdf/wamv_base.urdf.xacro
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] xacro: generating urdf/wamv_gazebo.urdf from urdf/wamv_gazebo.urdf.xacro
[  1%] xacro: generating worlds/buoyancy_plugin_demo.world from worlds/buoyancy_plugin_demo.world.xacro
[  2%] xacro: generating urdf/wamv_gazebo.urdf from urdf/wamv_gazebo.urdf.xacro
[  3%] xacro: generating urdf/wamv_base.urdf from urdf/wamv_base.urdf.xacro
[  3%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/wave_gazebo/world_models/ocean_waves/model.xacro
[  3%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/wave_gazebo/world_models/ocean_waves/model.xacro
[  4%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/wave_gazebo/world_models/ocean_waves/model.xacro
[  5%] creating dir /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/usv_gazebo_plugins/worlds
[  5%] creating dir /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/wamv_description/urdf
[  5%] creating dir /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/wave_gazebo/worlds
[  5%] Built target uuid_msgs_generate_messages_cpp
[  5%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/vrx_gazebo/models/robotx_2018_qualifying_avoid_obstacles_buoys/model.sdf
[  5%] creating dir /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/wamv_gazebo/urdf
[  5%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Gazebo.cc.o
[  5%] Built target _usv_msgs_generate_messages_check_deps_RangeBearing
[  5%] Built target trajectory_msgs_generate_messages_nodejs
[  5%] Built target sensor_msgs_generate_messages_cpp
[  5%] Built target trajectory_msgs_generate_messages_lisp
[  6%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/PhysicalConstants.cc.o
[  6%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Geometry.cc.o
[  6%] Built target tf_generate_messages_nodejs
[  6%] Built target std_srvs_generate_messages_nodejs
[  6%] Built target gazebo_msgs_generate_messages_cpp
[  6%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Physics.cc.o
[  6%] Built target std_srvs_generate_messages_lisp
[  6%] Built target std_srvs_generate_messages_eus
[  7%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Utilities.cc.o
[  7%] Built target std_srvs_generate_messages_cpp
[  7%] Built target dynamic_reconfigure_generate_messages_lisp
[  7%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Wavefield.cc.o

[  7%] Copying to devel space: /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/usv_gazebo_plugins/worlds/buoyancy_plugin_demo.world
[  7%] Built target usv_gazebo_plugins__xacro_auto_generate
[  7%] Built target tf2_msgs_generate_messages_cpp
[  7%] Built target gazebo_msgs_generate_messages_py
[  7%] Built target geometry_msgs_generate_messages_py
[  7%] Built target gazebo_ros_gencfg
[  7%] Built target roscpp_generate_messages_cpp
[  7%] Built target sensor_msgs_generate_messages_lisp
[  7%] Built target sensor_msgs_generate_messages_eus
[  7%] Built target usv_gazebo_plugins__xacro_auto_generate_to_devel_space_
[  7%] Built target roscpp_generate_messages_eus
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libsdformat9.so.9.6.0', needed by '/home/mrmccarr@ern.nps.edu/vrx_ws/devel/lib/libHydrodynamics.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[  8%] Building CXX object vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/WavefieldEntity.cc.o
[  8%] Built target sensor_msgs_generate_messages_py
[  8%] Built target tf2_msgs_generate_messages_eus
[  8%] Built target rosgraph_msgs_generate_messages_nodejs
[  8%] Built target actionlib_generate_messages_cpp
[  8%] Built target actionlib_msgs_generate_messages_py
[  8%] Built target geometry_msgs_generate_messages_lisp
[  8%] Built target tf_generate_messages_cpp
[  8%] Built target roscpp_generate_messages_nodejs
[  8%] Built target std_srvs_generate_messages_py
[  8%] Built target rosgraph_msgs_generate_messages_eus
[  8%] Built target tf_generate_messages_eus
[  8%] Built target rosgraph_msgs_generate_messages_lisp
[  8%] Built target tf_generate_messages_lisp
[  8%] Built target dynamic_reconfigure_generate_messages_cpp
[  8%] Built target geometry_msgs_generate_messages_cpp
[  8%] Built target dynamic_reconfigure_generate_messages_nodejs
[... 176 lines cut ...]

[ 56%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/vrx_gazebo/models/dock_2016_dynamic/model.sdf

[ 57%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/vrx_gazebo/models/dock_2022_dynamic/model.sdf
[ 57%] Generating /home/mrmccarr@ern.nps.edu/vrx_ws/src/vrx/vrx_gazebo/models/dock_2018_dynamic/model.sdf
[ 58%] Copying to devel space: /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/vrx_gazebo/worlds/wildlife_task.world
[ 58%] Copying to devel space: /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/vrx_gazebo/worlds/wind_test.world
[ 63%] Built target vrx_gazebo__xacro_auto_generate_to_devel_space_
[ 63%] Built target usv_msgs_generate_messages_eus
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:7948: vrx/vrx_gazebo/msgs/CMakeFiles/dock_placard_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:7975: vrx/vrx_gazebo/msgs/CMakeFiles/light_buoy_colors_msgs.dir/all] Error 2
[ 63%] Linking CXX shared library /home/mrmccarr@ern.nps.edu/vrx_ws/devel/lib/libball_shooter_plugin.so
[ 63%] Built target ball_shooter_plugin
make[1]: *** [CMakeFiles/Makefile2:1509: vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed

```

caguero, 2021-09-10:
Could you try removing your `build` and `devel` directories and recompile?

M1chaelM, 2021-09-10:
Yes! Thanks for the quick answer. It's now building again.

It looks like my system applied some updates over night, and these broke the build. `catkin_make` clean wasn't sufficient to clear it out. Sorry for the false alarm--with all the new changes to code I missed the fact that a new package had been installed in the background.

closed 2021-09-10 by M1chaelM