osrf/vrx #367: VRX Cloudsim real time factor < 0.7

M1chaelM, 2021-09-22:
VRX builds and runs correctly on our cloudsim image. However, the real time factor is low. Even running locally, I see RTF < 0.7

M1chaelM, 2021-09-22:
This could potentially be a problem with the Nvidia driver. The graphical "Additional Drivers" tab does not seem to be working, but command line tools report that the image is using the Nvidia driver:

[image]

This is the same driver as the host system, so I'm not yet sure if this is the problem. Since this image is relatively old now I may try rebuilding to see whether the issue persists with updated software.

M1chaelM, 2021-09-22:
@bsb808 When you try running locally could you let me know whether you are also still seeing a slow RTF?

bsb808, 2021-09-22:
@M1chaelM - I can try to repeat what you are seeing, but it sounds like we are seeing consistent low performance both running on AWS and running locally.  This seems to suggest a problem with the image.   Have you tried this with both the matlab and non-matlab image?

M1chaelM, 2021-09-24:
To troubleshoot this I compared 4 different scenarios:
* Running our cloudsim image locally and accessing through the browser using novnc
* Running our cloudsim image locally without novnc, using x11 directly as we do with our dockwater image
* Running our dockwater image (locally with x11)
* Running our dockwater image through novnc, using `rocker --nvidia --novnc --turbovnc --user --user-override-name=developer noetic:latest`
* Running on the "bare metal" using a host install

Since the default vrx.launch demo has a lot of floating objects, including 3 moving animals, I also ran each scenario using the simpler station_keeping.launch file. 

|   | cloudsim novnc local | cloudsim x11 | dockwater novnc local | dockwater x11   | bare metal  |
|---|---|---|---|---|---|
|vrx.launch   | 0.59  | 0.75  | 0.59 | 0.98 | 0.98  |
| station_keeping.launch  |  0.59 | 0.86  | 0.59 | 0.98   | 0.98  |

## Takeaways
* Using NoVNC seems to produce a major performance hit.
* I don't know why we didn't notice this before. Either NoVNC has changed or VRX is much more demanding than VORC, or we were just not very observant. Unfortunately I could not get VORC working or find a previous state of the repository where it would build.
* I will need to work with @tfoote and @nkoenig to see whether there is anything we can do about this.

M1chaelM, 2021-09-24:
One more note: I get a slight performance boost if I shrink the size of the browser tab. This bumps the RTF from about 0.59 to about 0.65.

bsb808, 2021-10-05:
Some additional data from testing.

=== VORC ===

From December, we know that VORC ran at RTF ~ 1.0.   
Currently VORC needs some additional work to operate with the full new VRX setup, but running with the current master branches, we see...

```
roslaunch vorc_gazebo marina.launch
```

* RTF ~ 0.99, FPS ~10.  

[image]

=== VRX ===

```
roslaunch vrx_gazebo vrx.launch 
```

* RTF ~ 0.4, FPS ~ 3.

[image]

caguero, 2021-10-05:
This is the pull request that enabled reflections/refractions: https://github.com/osrf/vrx/pull/361/files

If this is the root of the problem we'll need a way to enable/disable it but for a quick test you can edit the `#ifdef`.

JonathanWheare, 2021-10-10:
Hi everyone,

I believe I may have the same issue.  When running the VRX simulation on my laptop using the Sydney Regatta site, I had a real time factor of approximately 0.8, and the gzserver process was using around 130% CPU.  Deleting the sydneyregatta model brought performance back up to the expected level.
I found that disabling the collision tag for the Sydney Regatta model appeared to greatly improve performance, returning the simulation to a real-time factor of 1.0 with the gzserver process only using about 80% CPU.
Could it be the collision model?

Regards,
Jonathan.

caguero, 2021-10-12:
As a workaround until we fix it properly, you can comment this block:

https://github.com/osrf/vrx/blob/master/vrx_gazebo/models/sydney_regatta/model.sdf#L104

bsb808, 2021-10-15:
I believe we moved from g3.4xlarge to a g3.8xlarge for cloudsim.  I did a quick test of the VORC marina.launch scenario and it appears to have the same performance:
RTF ~ 1.0 and FPS ~ 11.0

I’ll try to look at it again today, but this may suggest that the extra cost isn’t resulting in better performance.

caguero, 2025-07-25:
Closing this one as VRX cloudsim is no longer active.

closed 2025-07-25 by caguero