osrf/vrx #370: VRX Crashes Every Couple Minutes

ayarst, 2021-09-27:
No matter which launch file I use, VRX crashes after a couple minutes and I have to restart everything. It is really starting to affect my productivity. These are the commands I'm running:

roslaunch vrx_gazebo scan_and_dock.launch urdf:=$HOME/my_wamv/my_wamv.urdf

roslaunch wamv_gazebo localization_example.launch

roslaunch wamv_gazebo rviz_vrx.launch

Every several minutes, the first terminal (vrx) errors out with:

Segmentation fault (core dumped)
[gazebo-2] process has died [pid 47690, exit code 139, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/robotx/vrx_ws/devel/share/vrx_gazebo/worlds/dock.world __name:=gazebo __log:=/home/robotx/.ros/log/64c56d8e-1fa9-11ec-9a4e-c99f73aa7815/gazebo-2.log].
log file: /home/robotx/.ros/log/64c56d8e-1fa9-11ec-9a4e-c99f73aa7815/gazebo-2*.log

I would love to know if it is user error or if it is something that needs to be fixed. 

Many thanks.

caguero, 2021-09-27:
Hi @ayarst , this is probably expected because you're running a world with a scoring plugin that is going to terminate the world after 5 mins of `running` mode. I recommend running `vrx.launch` or modify the `<running_state_duration>300</running_state_duration>` in the world file. 

By the way, `scan_and_dock` is from the 2019 competition. The 2022 world is `scan_dock_deliver`.

ayarst, 2021-09-27:
@caguero thank you so much. I will look into that immediately! Really appreciate the quick response time.

ayarst, 2021-09-27:
@caguero while I have you, I actually have one other question. Where exactly is wamv/base_link located? I am trying to set up my sensors_config.yaml and without knowing the location of base_link relative to our robot frame, I have to eyeball it and I do not want to do that.

caguero, 2021-09-27:
Can't you see the `wamv/base_link` frame link in `rviz`? I'm not sure if I can give you a precise location as this is the main origin of the model.

ayarst, 2021-09-27:
Yeah I can, was just wondering if there was something more you could give me. No worries, I'll figure it out. I really appreciate your help.

closed 2021-09-27 by ayarst