osrf/vrx #372: map ballshooter launch command to joystick button

M1chaelM, 2021-10-05:
It would be convenient for testing to make it possible for someone using gamepad teleop to be able to fire the ballshooter by pressing a button on the gamepad. In the past we have done something similar on other projects.

M1chaelM, 2021-10-05:
@bsb808 Could you point us to an example of a project where we've done this before?

j-herman, 2021-11-03:
@M1chaelM 
I've got a working version of this on branch jessica/ball_shooter_teleop, in case you want to try it.  I won't put in a pull request until after 1.5 is out of the way.  
To test: 
Run scan/dock/deliver task 
`roslaunch vrx_gazebo scan_dock_deliver.launch`
Run teleop node
`roslaunch vrx_gazebo usv_joydrive.launch`
To confirm the ball shooter actually fires, listen on the appropriate topic
`rostopic echo /wamv/shooters/ball_shooter/fire`
Hit the left lower trigger and you should see the ball shoot, as well as an empty message (---) on the firing topic.

closed 2021-11-15 by j-herman