osrf/vrx #373: Add the ability to configure the ball shooter

caguero, 2021-10-05:
In a similar way the teams can choose their sensor and thruster configurations, the teams should be able to specify where they want the ball shooter to be located in the WAM-V.

* [ ] We'll need to update the submission wiki pages. See [phase 2](https://github.com/osrf/vrx/wiki/vrx_2022-phase2_dress_rehearsal) and [phase 3](https://github.com/osrf/vrx/wiki/vrx_2022-phase3_challenge) instructions.
* [ ] Update the ["Creating a custom WAM-V thruster and sensor configuration for competition"](https://github.com/osrf/vrx/wiki/tutorials-Creating%20a%20custom%20WAM-V%20Thruster%20and%20Sensor%20Configuration%20For%20Competition) tutorial.
* [ ] Update [vrx-docker] [prepare_team_wamv.bash](https://github.com/osrf/vrx-docker/blob/master/prepare_team_wamv.bash) for including the `ball_shooter.yml` file.
* [ ] Update [vrx] [vrx_gazebo/src/vrx_gazebo/configure_wamv.py](https://github.com/osrf/vrx/blob/master/vrx_gazebo/src/vrx_gazebo/configure_wamv.py)
* [ ] Update [vrx] [compliance.py](https://github.com/osrf/vrx/blob/master/vrx_gazebo/src/vrx_gazebo/compliance.py)

I think these are the places to be modified.

crvogt, 2021-11-10:
@caguero @M1chaelM  I was thinking of going through the tutorials and changing sensor references to component and adding a line introducing the "component" umbrella term. Do you have an opinion on this?

caguero, 2021-11-10:
It sounds like a good idea to me, especially because it might surprise previous users about the change. Thanks!

closed 2021-11-16 by crvogt