osrf/vrx #395: Wiki Update: acoustic_pinger_tutorial

Tinker-Twins, 2021-11-07:
[Reading acoustic beacon messages](https://github.com/osrf/vrx/wiki/acoustic_pinger_tutorial#reading-acoustic-beacon-messages) section of the `acoustic_pinger_tutorial` seems outdated. Paticularly, the launch command to bring up the VRX 2019 simulation environment needs to be updated since the `vrx_2019.launch` file is no longer in the `vrx_gazebo` package, but instead in the `vrx_2019` package (with a different name, i.e. `vrx.launch`). The correct launch command (according to the updated repository structure) should therefore be as follows:
```bash
$ roslaunch vrx_2019 vrx.launch
```

Additionally, in order to reproduce results similar to those described in [Visualize sensor measurements in RViz](https://github.com/osrf/vrx/wiki/acoustic_pinger_tutorial#visualize-sensor-measurements-in-rviz) section of the same Wiki, the pinger position needs to be set to `[65.0, -47.5, -2]`. Pull request #394 has been created with updated `pinger.yaml` file to include these coordinates. A hint to comment/uncomment these coordinates may be included in the tutorial Wiki.

Tinker-Twins, 2021-11-08:
In accordance with recent changes to pull request #394, the hint to comment/uncomment pinger coordinates (as mentioned in [earlier comment](https://github.com/osrf/vrx/issues/395#issue-1046813876)) is no longer required. However, the first launch command under section [Visualize sensor measurements in RViz](https://github.com/osrf/vrx/wiki/acoustic_pinger_tutorial#visualize-sensor-measurements-in-rviz) of the tutorial Wiki needs to be updated.

From:
```bash
$ roslaunch vrx_gazebo pinger.launch
```
To:
```bash
$ roslaunch vrx_2019 pinger.launch
```

caguero, 2021-11-08:
Thanks! The wiki has been updated.

caguero, 2021-11-08:
See https://github.com/osrf/vrx/pull/394

closed 2021-11-08 by caguero