osrf/vrx #398: Documentation Error: Thruster Configuration

Tinker-Twins, 2021-11-16:
Section 3 of the [Tecnical Guide](https://robonation.org/app/uploads/sites/2/2021/09/VirtualRobotX2022_Technical-Guide_v1.0.pdf) elucidates the `Propulsion Configuration` (pages 2-3) wherein `Figure 2: Allowed Thruster Regions in WAM-V` and `Table 1: Dimensions of Bounding Boxes` do not seem to comply with each other.

Particularly, the bounding boxes shown in the figure have larger dimensions along `x` and `y` axes as compared to the `z` axis. However, the table states otherwise, and needs to be updated. The correct dimensions for `Size (x y z)` of bounding boxes in the table (column 3) seems to be:
- **Back Left:** `2 2 1.2`
- **Back Right:** `2 2 1.2`
- **Front Left:** `2 2 1.2`
- **Front Right:** `2 2 1.2`
- **Center:** `5 2 1.2`

With this two fold increase in `x` and `y` dimensions of the bounding boxes (to cover either sides of their center), the figure and table seem to agree with each other.

caguero, 2021-11-19:
This is the visualization of the allowed thruster areas with the current VRX parameters:

[image]

I'd say we shouldn't modify `x` and `y`. Maybe we could reduce `z` potentially.

Tinker-Twins, 2021-11-19:
Here's the exact image from the [documentation](https://robonation.org/app/uploads/sites/2/2021/09/VirtualRobotX2022_Technical-Guide_v1.0.pdf):

[image]

It seems that the bounding boxes in this figure are either outdated, or they are submerged and hence their portions below the water-level aren't visible. The figure in the [earlier comment](https://github.com/osrf/vrx/issues/398#issuecomment-974298389) seems to agree with the ratios of 1:1.2 for `y:z`, so it would be a good idea to update the image in the documentation so as to avoid any confusion.

Tinker-Twins, 2021-11-19:
**How are the `Center` and `Size` parameters of the bounding boxes supposed to be interpreted?**

[image]

Consider the above table from page 3 of the [documentation](https://robonation.org/app/uploads/sites/2/2021/09/VirtualRobotX2022_Technical-Guide_v1.0.pdf). For example, for the `y` coordinate tolerance of `Back Left` thruster, there are two possible ways to interpret the `Size` parameter:

- **Case 1:** `Size` is measured in both directions about the `Center` [i.e. 1m `Center` +/- 1m `Size`] corresponding to a total tolerance of 2m. Refer the following figure:
[image]

- **Case 2:** `Size` is measured as the total dimension of the side of the bounding box [i.e. 1m `Center` +/- 0.5m `Size`] corresponding to a total tolerance of 1m. Refer the following figure:
[image]

From the default `X` thruster configuration described in the [wamv_x_thrusters.xacro](https://github.com/osrf/vrx/blob/43de9607a5fc3cfc08a802c24dd89de0ddbc10da/wamv_gazebo/urdf/thruster_layouts/wamv_x_thrusters.xacro#L6) file, it can be observed that the `Front Left` thruster is located at `1.6 0.7  0.25`.

- If you are comfortable with `Case 1` representation, the table values need not be updated, but it would be a good idea to include the [respective image](https://user-images.githubusercontent.com/40835293/142700352-237a18ba-0d0b-4561-be55-285b07a8212a.png) in the documentation to avoid any confusion.

- If you are comfortable with `Case 2` representation, the table values need to be doubled in all 3 dimensions, and it would be a good idea to include the [respective image](https://user-images.githubusercontent.com/40835293/142700332-043ac264-afb2-44a2-8daa-f7fdd37d5b11.png) in the documentation to avoid any confusion.

caguero, 2021-11-22:
Case 2 is the correct interpretation of the geometry of the bounding box. I don't think we need to double the values in all three dimensions, as the graphical representation looks correct to me. I see what you're saying about the `Front Left` and `Front Right` thrusters.

Would it be OK for you if we update the position of the front thrusters to `1.2 1 0 0 0 0` and `1.2 -1 0 0 0 0` respectively. That way, the default `X` configuration is within the limits.

```

```

Tinker-Twins, 2021-11-23:
Yes, updating pose of the `Front Left` and `Front Right` thrusters to `1.2 1 0 0 0 0` and `1.2 -1 0 0 0 0` respectively resolves the issue.

The updated `Center` and `Size` parameters for the truster configuration bounding boxes are as follows. The same have been verified in Gazebo.

| Thruster Name | Bounding Box Center (x y z R P Y) | Bounding Box Size (x y z) |
| -------------------- | ---------------------------------------------- | ----------------------------------- |
| Front Left          | 1.2 1 0 0 0 0                                    | 1 1 1.2                                |
| Front Right        | 1.2 -1 0 0 0 0                                  | 1 1 1.2                                |
| Center               | 0.25 0 0 0 0 0                                 | 2.5 1 1.2                             |
| Back Left           | -2.25 1 0 0 0 0                                | 1 1 1.2                                |
| Back Right        | -2.25 -1 0 0 0 0                               | 1 1 1.2                                |

In addition to updating the table values, kindly update the figure in the [documentation](https://robonation.org/app/uploads/sites/2/2021/09/VirtualRobotX2022_Technical-Guide_v1.0.pdf) with the following image, since the original figure is sligtly misleading as mentioned in [earlier comment](https://github.com/osrf/vrx/issues/398#issuecomment-974501089).

[image]

Tinker-Twins, 2021-11-25:
Could we update pose of the `Front Left` and `Front Right` thrusters to `1.25 1 0 0 0 0` and `1.25 -1 0 0 0 0` respectively so that front and rear trusters could be placed equidistant (w.r.t. `x` axis) from the origin of WAM-V frame?

The updated `Center` and `Size` parameters for the truster configuration bounding boxes will be as follows. The same have been verified in Gazebo.

| Thruster Name | Bounding Box Center (x y z R P Y) | Bounding Box Size (x y z) |
| -------------------- | ---------------------------------------------- | ----------------------------------- |
| Front Left          | 1.25 1 0 0 0 0                                  | 1 1 1.2                                |
| Front Right        | 1.25 -1 0 0 0 0                                | 1 1 1.2                                |
| Center               | 0.25 0 0 0 0 0                                 | 2.5 1 1.2                             |
| Back Left           | -2.25 1 0 0 0 0                                | 1 1 1.2                                |
| Back Right        | -2.25 -1 0 0 0 0                               | 1 1 1.2                                |

caguero, 2021-11-25:
Sure, #405 has been updated.

closed 2021-12-07 by Tinker-Twins