osrf/vrx #399: Wind parameter documentation

bsb808, 2021-11-17:
I may be missing where this is, but in the Technical Guide we link to the [Theory of Operations wiki](https://github.com/osrf/vrx/wiki/vrx_theory).  In general, it would be better to use the OCEANS paper we wrote to document the underlying models used in the plugins.  

Specifically, I don't see where the meaning of the wind plugin parameters is described, e.g.
```
direction="270"
      mean_vel="0"
      var_gain="0"
      var_time="10"
      seed="42">
```

M1chaelM, 2021-11-22:
@bsb808 I suspect this just hasn't been updated since 2019, so it still points to your old Theory of Operations document, which is linked from the Theory of Operations wiki page: https://github.com/bsb808/robotx_docs/blob/master/theoryofoperation/theory_of_operation.pdf

I think that document eventually became the paper and is now obsolete. If that's the case I can update the link to point at our paper instead.

Since the section on wind modeling in the paper poses a fairly high barrier of entry, we also have a Wiki page that walks through examples of how to change the wind parameters: https://github.com/osrf/vrx/wiki/changing_wind_params_tutorial

That page could probably use some cleaning up and it would really be nice if it linked to the paper and said specifically which of the various variables we use in the paper correspond with which parameter names in the code.

Finally, all the tutorials for changing the environment parameters are listed under the tutorial on [changing model and environmental parameters](https://github.com/osrf/vrx/wiki/changing_plugin_params_tutorial), which is under the [Customizing VRX](https://github.com/osrf/vrx/wiki/customizing_vrx_tutorials) section of the wiki. These are admittedly nested a bit deep because I assumed they wouldn't be of immediate interest to most teams. Can you tell me where you'd expect to find them? I could add a link from the Theory of Operations Wiki, for example.

M1chaelM, 2022-06-13:
@caguero I propose we address this as part of our re-organization of the documentation for VRX in general.