osrf/vrx #403: Clean up example worlds and configs

M1chaelM, 2021-11-22:
## Problem
Changes to VRX have caused our examples to get out of sync with those provided or generated by the VRX repository. This is a little confusing.

## Proposal
* [ ] Remove example_team_2 from vrx_docker. These teams only have very slightly differing wamv configs. Unless we have a tutorial that requires them, one should be enough.
* [ ] Generate worlds described by the task_config yaml files in vrx-docker and cache them in the vrx-docker repository.
* [ ] Replace the worlds in vrx under 2022_practice so they are identical to those generated by the vrx-docker yaml. Currently they are almost identical but have a number of trivial variations (inlcuding elements appearing in a different order). This makes it hard to know whether they are really the same or not without a detailed inspection.
* [ ] Replace the sensor_config.yaml file in the example_team in vrx-docker with a copy of our new example_component_config.yaml, which contains the new ballshooter.

M1chaelM, 2021-11-22:
@caguero and @crvogt Could you review the plan above and make sure I'm not eliminating something we need?

M1chaelM, 2021-11-22:
Apologies, I forgot we updated the component_config.yaml for both example teams in https://github.com/osrf/vrx-docker/pull/42. The updated examples are different enough that it seems to justify having two of them.

M1chaelM, 2021-11-22:
On further reflection, I don't think we need to cache the generated worlds in the vrx-docker repository since they are already in the vrx repository. So I will just update the vrx_2022 versions so they are exact matches.

caguero, 2021-11-25:
See #404 .

closed 2021-11-25 by caguero