osrf/vrx #406: Test the Phase 1 submission instructions:

M1chaelM, 2021-11-23:
https://github.com/osrf/vrx/wiki/vrx_2022-phase1_hello_world

It's not necessary to follow every link, but testing should also include looking at the Submission Process tutorial referenced at the bottom: https://github.com/osrf/vrx/wiki/submission_process

j-herman, 2021-11-29:
Ran through the submission process.  If the pull request in the vrx-events repo contains what you are looking for, then the test was successful.  Two things that should be fixed:
The directory structure referenced in the submission process tutorial doesn't yet exist for 2022 and needs to be created.
The link to the VRX task description document goes to the documentation page, which does not contain the task description document.  Tech guide links throughout the tutorials do the same.

M1chaelM, 2021-11-29:
Thanks! Very helpful.
> * The directory structure referenced in the submission process tutorial doesn't yet exist for 2022 and needs to be created.
This is addressed with [PR 1 of that repository](https://github.com/osrf/vrx-events/pull/1) but it looks like we forgot to merge it. Hopefully we'll get this done today.
> * The link to the VRX task description document goes to the documentation page, which does not contain the task description document.  Tech guide links throughout the tutorials do the same.
I updated the text and link to the task description document. Can you point me to the tech guide links you're thinking of?

j-herman, 2021-11-29:
All is good with your change, I think - I don't remember exactly where all the other links were coming from, but since they all pointed to the same documentation page they should all be fixed now.  New PR looks good and the example submissions will be helpful in clarifying expectations.  We can close this if you're ready.

M1chaelM, 2021-11-29:
@j-herman Both of the issues you raise above should now be addressed. I'm not seeing your PR in the vrx-events repo though. What was the PR #?

j-herman, 2021-11-29:
Whoops!  Forgot to click submit... you should see it now.

M1chaelM, 2021-11-29:
OK, thanks! I just reviewed it. The only thing I want to check on is the directory structure you used. See comments: https://github.com/osrf/vrx-events/pull/4#pullrequestreview-818364943

M1chaelM, 2021-11-30:
Closing with close of PR: https://github.com/osrf/vrx-events/pull/4

closed 2021-11-30 by M1chaelM