osrf/vrx #409: VRX pinger launch failure on Ubuntu 18.04 on ROS melodic

ofirc, 2021-11-27:
Hi,
I installed ROS melodic on Ubuntu 18.04 by following this guide:
http://wiki.ros.org/melodic/Installation/Ubuntu

And then used the following commands to build and launch VRX:
```
# Install the prerequisites for VRX
sudo apt install python3-pip
pip3 install numpy

# Create a workspace
mkdir -p ~/vrx_ws/src
cd ~/vrx_ws/src

# Clone the Git repository to your local workspace
git clone https://github.com/osrf/vrx
cd ~/vrx_ws

# Install ROS dependencies
rosdep update
rosdep install --from-paths src --ignore-src -r -y

# Build the VRX package
catkin_make

# Setup the environment such that roslaunch finds the vrx_gazebo package.
source ~/vrx_ws/devel/setup.bash

# Launch gazebo and other ROS dependencies.
roslaunch vrx_gazebo vrx.launch
```

Now, the Gazebo window is opening and I see the ASV and everything, but the pinger doesn't launch correctly:
```
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
ModuleNotFoundError: No module named 'rospkg'
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ModuleNotFoundError: No module named 'rospkg'
[ INFO] [1638027771.823364800]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1638027771.831318100]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[set_pinger_position-5] process has died [pid 8475, exit code 1, cmd /home/ofir/vrx_ws/src/vrx/vrx_gazebo/nodes/set_pinger_position.py __name:=set_pinger_position __log:=/home/ofir/.ros/log/ace58ed6-4f98-11ec-a9fa-00155d087f03/set_pinger_position-5.log].
log file: /home/ofir/.ros/log/ace58ed6-4f98-11ec-a9fa-00155d087f03/set_pinger_position-5*.log
[pinger_visualization-6] process has died [pid 8476, exit code 1, cmd /home/ofir/vrx_ws/src/vrx/vrx_gazebo/nodes/pinger_visualization.py __name:=pinger_visualization __log:=/home/ofir/.ros/log/ace58ed6-4f98-11ec-a9fa-00155d087f03/pinger_visualization-6.log].
log file: /home/ofir/.ros/log/ace58ed6-4f98-11ec-a9fa-00155d087f03/pinger_visualization-6*.log
```

Any idea what's wrong?

BTW, I have rospkg installed:
```
ofir@ocohen11-MOBL:~$ which python
/usr/bin/python
ofir@ocohen11-MOBL:~$ python -c 'import rospkg'
ofir@ocohen11-MOBL:~$ python --version
Python 2.7.17
ofir@ocohen11-MOBL:~$
```
so I either missed a step or something is messed up with VRX environment.

Thanks,
Ofir

ofirc, 2021-11-27:
I was able to work-around it by hacking the launcher.py file:
`sudo vim /opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py`

and then adding:
```
if '/usr/lib/python2.7/dist-packages' not in sys.path and '/usr/lib/python2.7/dist-packages/' not in sys.path:
    sys.path.append('/usr/lib/python2.7/dist-packages')

```
But it's not a permanent solution :-)

Looking forward to hearing the root cause.

caguero, 2021-12-03:
I'm sorry but we no longer support Ubuntu 18.04. See the [system requirements](https://github.com/osrf/vrx/wiki/system_requirements). I know this doesn't answer your question but I'd recommend to update to Ubuntu 20.04 and I guess you won't have this issue anymore.

closed 2021-12-03 by caguero

ofirc, 2021-12-03:
Thanks @caguero good to know :-)