osrf/vrx #412: Update Phase 2 testing tutorial for 2022

M1chaelM, 2021-12-07:
In 2019, we provided this tutorial for phase 2 testing: https://github.com//osrf/vrx/wiki/vrx_2019-phase2_testing

We need to make a 2022 version using the new example worlds. 

In the future, we should tie this tutorial to the PR in which we create the example worlds.

M1chaelM, 2021-12-09:
@caguero our [Phase 2 - Dress Rehearsal](https://github.com/osrf/vrx/wiki/vrx_2022-phase2_dress_rehearsal) says: 

"For the evaluation of your Phase 2 submissions, the solutions will be run through each of these example worlds, along with some new configurations that are not released prior to the Phase 2 deadline."

Is this still correct?

M1chaelM, 2021-12-09:
The Phase 2 practice tutorial for 2019 was extremely long. To make it more manageable, I've separated it into a general opening and 6 subpages describing the tasks. I fixed up the opening but I could use help with the subpages, which are here:
* [ ] [Task 1: Stationkeeping Practice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task1_practice.md)
* [ ] [Task 2: Wayfinding Practice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task2_practice.md)
* [ ] [Task 3: Perception Practice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task3_practice.md)
* [ ] [Task 4: Wildlife Practice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task4_practice.md)
* [ ] [Task 5: Gymkhana Practice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task5_practice.md)
* [ ] [Task 6: Scan and Dock and DeliverPractice Worlds](https://github.com/osrf/vrx/wiki/vrx_2022-task6_practice.md)

Each page currently contains the content we had in 2019, and it needs to be updated to the correct content for 2022. @crvogt and @j-herman , do you have some time to help with this?

M1chaelM, 2022-01-31:
> @caguero our [Phase 2 - Dress Rehearsal](https://github.com/osrf/vrx/wiki/vrx_2022-phase2_dress_rehearsal) says:
>
> "For the evaluation of your Phase 2 submissions, the solutions will be run through [...]

Let's discuss this in our meeting on 2/1.

caguero, 2025-05-08:
Closing this issue as it was tracking a previous development cycle.

closed 2025-05-08 by caguero