osrf/vrx #413: Outdated wiki: changing_plugin_params_tutorial

AntonBock, 2021-12-08:
Hello

I am trying to add fog to my scene. According to [this tutorial](https://github.com/osrf/vrx/wiki/changing_plugin_params_tutorial), the file location for changing fog parameters is in `vrx_gazebo/worlds/sandisland.xacro`. However, i cannot find this file. 

I don't know if I am missing something super obvious, or if a file location has changed since the last update of the wiki. Hope someone can help!

Best regards, 
Anton

AntonBock, 2021-12-08:
If anyone can point me in the correct direction of adding fog to a scene, that would be wonderful as well

M1chaelM, 2021-12-09:
You're right--thanks for pointing this out. I've updated the tutorial. The `sandisland.xacro` file has been moved to the `vrx_2019` package. The VRX 2022 environment uses `sydneyregatta.xacro` instead, so if you are working on VRX 2022 that is the one you want.

AntonBock, 2021-12-10:
No problem, thanks! 

Can you elaborate on how specifically the fog is added? I have been through the documentation and the referenced files, and cannot figure it out.

M1chaelM, 2021-12-12:
Sure. I've made some additional updates to the tutorial to try to clarify the process: https://github.com/osrf/vrx/wiki/changing_visual_params_tutorial 

Could you check whether this resolves the problem for you? If not, it would be helpful to know what you've tried and what the results are.

AntonBock, 2021-12-13:
Thank you! I was missing the `type` tag, adding this fixed it.

closed 2021-12-13 by AntonBock