osrf/vrx #414: VRX_Gazebo scoring plugins errors

mikedef, 2021-12-20:
I am getting an error due to the following when building the code on an Ubuntu 20.04 laptop.
  fatal error: ignition/utilities/SuppressWarning.hh: No such file or directory
Is there any way around the error besides not building any of the scoring plugins?

caguero, 2021-12-20:
Hi @mikedef , could you confirm your OS, Gazebo and ROS versions?

mikedef, 2021-12-20:
Ubuntu 20.04, Gazebo 11, ROS noetic

M1chaelM, 2022-01-07:
Hi @mikedef . We have not able to reproduce this error. Are you building using the instructions given in the setup tutorial: https://github.com/osrf/vrx/wiki/system_setup_tutorials ?

If not, could you send the commands you're running?

caguero, 2022-01-07:
As a sanity check, could you also update the system doing:

```
sudo apt-get update
sudo apt-get upgrade
```

mikedef, 2022-01-21:
It still does not work on my laptop. I did update and upgrade but it does not work. I did install it on my desktop and it works fine. So maybe something particular to my laptop. If this is not reproducible then I would close this issue. Thanks for the help!
Mike

closed 2022-01-21 by mikedef