osrf/vrx #415: Issue loading VRX environment

FrancisGoncalves, 2021-12-23:
Hi, we are having some issues with some parts of the environment not loading properly. Some trees and other landscapes are not loading correctly.
Also, when visualizing the camera images, we noticed that the horizon becomes dark (sea, shore, buildings) and makes things difficult to recognize at a certain distance. ​​ ​​
Is anyone else facing these problems?

Thank you
[image]

M1chaelM, 2022-01-07:
@FrancisGoncalves Could you post your hardware configuration? It looks like there is a problem with the rendering, but we have not yet been able to reproduce it.

FrancisGoncalves, 2022-01-07:
@M1chaelM We are running Ubuntu 20.04 dual booted with windows. 
We are using a Precision 7820 Tower with an Intel Xeon Platinum 8160 CPU.
We have NVIDIA Quadro P4000 graphics card and 192 GB of RAM.
We allocated only 40 GB of storage for Ubuntu.
Let me know if you need any other information.
Thank you

caguero, 2022-01-07:
Do you have the Nvidia drivers running?

FrancisGoncalves, 2022-01-07:
@caguero Thank you! We were using Nouveau display driver. 
We have now changed to one of the eight NVIDIA drivers options. The environment now loads with no issues. However, we are getting a very low frame rate (around 3 fps), while previously we were getting 30 fps. 
Is there a specific NVIDIA driver we should select?

colejames-wsu, 2022-01-11:
We have turned off UEFI secure boot option and the environment is now loading properly and running at 60 fps. 
Thank you for your help.

caguero, 2022-01-12:
Good to know!

closed 2022-01-12 by caguero