osrf/vrx #426: Bearing calculation for acoustic pinger does not seem to be correct

andrew-aj, 2022-02-02:
For the acoustic pinger the calculation for the bearing does not seem to be correct. For example, with a bearing value of 0.22 when we yaw our boat right 0.22 radians the new bearing value is 0.44 whereas it should be 0. It seems like the bearing calculation is reversed and is counter-clockwise when bearing normally is calculated with positive being clockwise.

acxz, 2022-07-13:
I fully agree with @andrew-aj here, bearing is calculated clockwise from north.

From [wikipedia](https://en.wikipedia.org/wiki/Bearing_(angle)): 

> In nautical [navigation](https://en.wikipedia.org/wiki/Navigation) the absolute bearing is the [clockwise](https://en.wikipedia.org/wiki/Clockwise) angle between [north](https://en.wikipedia.org/wiki/North) and an object observed from the vessel.

j-herman, 2022-07-25:
VRX defines the pinger bearing in the body frame of the WAMV, documented in the wiki: https://github.com/osrf/vrx/wiki/frame_conventions

acxz, 2022-07-25:
Thank you for taking a look at this issue! I don't think we have a disagreement on what frame the bearing is computed with respect to, but rather how VRX defines bearing.

Typically, it is considered a clockwise angle in the body frame. Would it be possible to update the documentation to clarify that the bearing is defined as counterclockwise in the body frame?

I think that would be sufficient to close this issue and to prevent confusion for future users familiar with the clockwise definition of the bearing.

An addition like this at the end of the first paragraph under https://github.com/osrf/vrx/wiki/frame_conventions#acoustic-pinger would go a long ways:
`The bearing is computed counterclockwise from the body frame x-axis.`

j-herman, 2022-07-25:
@acxz Thanks for the comment!  There was a lot of discussion when this issue first came up as to whether we should make the change.  When we decided to stay with the REP, we added the wiki page to try to help with just that confusion - sounds like a good recommendation to make this easier to interpret for everyone.