osrf/vrx #431: Some examples use wind parameters outside the specified envelope

M1chaelM, 2022-02-22:
Our documentation specifies that the value of `wind_variance_time` should be between 2 and 20. However, we have some practice example worlds in which it is set to 1. This could be misleading, so we should update them so they are at least 2.

M1chaelM, 2022-02-22:
This is partially addressed by PR #432, but needs to also be addressed upstream in the yaml files in the `vrx-docker` repository.

j-herman, 2022-03-18:
@M1chaelM I have made updates to all the yamls in the phase2 repo - I can make the changes to these files now, or once we add the phase2 worlds as additional practice.

M1chaelM, 2022-03-31:
@j-herman I think the `vrx-docker` side of this was addressed in https://github.com/osrf/vrx-docker/pull/44 so we may be able to close this now. Is there anything else to do?

closed 2022-03-31 by j-herman