osrf/vrx #440: WAMV can move in the perception task trials

M1chaelM, 2022-03-23:
When testing our example images on the perception task in PR #48 we found that the WAMV was not locked in the perception tasks and could respond to thrust commands. So it seems like the "locked" variable is either not set properly or is being ignored. It could be that this is a consequence of the way we are generating worlds from `yaml`.

M1chaelM, 2022-03-23:
This issue does not occur in the perception example. After running
```
roslaunch vrx_gazebo perception_task.launch
```
the following commands have no effect:
```
RATE=1
CMD=2
rostopic pub /wamv/thrusters/left_thrust_cmd std_msgs/Float32 -r ${RATE} -- ${CMD} &
rostopic pub /wamv/thrusters/right_thrust_cmd std_msgs/Float32 -r ${RATE} -- ${CMD}
```

M1chaelM, 2022-03-23:
This is also not a problem when using the example worlds following [the instructions here](https://github.com//osrf/vrx/wiki/vrx_2022-phase2_practice) in which we explicitly set the wamv_locked variable to true when we launch the trial. 

So it could be that this problem is isolated to our competition testing apparatus. I see that the `run_vrx_trial.sh` script we call in the `vrx-docker` server is using the variable `robot_locked` instead of `wamv_locked` and this doesn't seem to make any sense because `vrx.launch` does not have a corresponding variable. However, when I attempt to rename to `wamv_locked` and rebuild, the problem persists, so I must still be missing a piece.

M1chaelM, 2022-03-26:
@caguero Could you take a look? I may be missing something obvious.

caguero, 2022-03-29:
@M1chaelM , that's the issue, the parameter should be `wamv_locked`. Did you rebuild the server Docker image after fixing the parameter?

alexperez33, 2022-04-06:
I am not sure if this issue is still here, but I think my issue #445 may be relevant.

M1chaelM, 2022-04-09:
Resolved (for now) with #446 and https://github.com/osrf/vrx-docker/pull/49

closed 2022-04-09 by M1chaelM