osrf/vrx #442: Ocean current?

maghauke, 2022-03-25:
Hi, does the simulator simulate ocean current or is it only a visual effect that the water-surface moves if waves and wind are off?

caguero, 2022-03-29:
We don't simulate ocean current. The wave direction is configurable, so I guess it would be cool to align it with the wind vector but out of scope in the short term unless someone volunteer. Thanks for the idea!

closed 2022-03-29 by caguero

bsb808, 2022-03-29:
Just a note for posterity on this issue.   It was a intentional not to include ocean current as an an external disturbance in the simulation.  The reason is that for vessels such as the WAM-V (vessels with small draft and large windage), wind is a much stronger influence than water current.  Having independent wind and wave disturbances provides a rich envelope for exercising motion control, so adding water current would be redundant.

From an application perspective when the wind and seas are aligned, the control problem is a bit easier - the system typically will point the bow into the stern.  It is actually more challenging (and often the case in the field) that the wind and seas are not aligned, which makes station keeping a more challenging problem.

caguero, 2022-03-29:
Thanks for the clarification Brian and in case I wasn't clear, I meant to align the visual aspect of the wave with the wind, so nothink should change from the control point of view, only adding some bling to the simulation :)