osrf/vrx #445: Unable to release joints in "ready state" because they do not exist

alexperez33, 2022-04-01:
When running any of the tasks, I noticed that I would always get the error ```Unable to release [wamv_external_pivot_joint]``` and ```Unable to release [wamv_external_riser]```. I also noticed that my urdf of the boat does not have these joints, which is generated using the generate_wamv.launch script. This seems to cause the boat to not be locked in place during the initial state of a task. This is especially problematic during tasks with big waves and wind because we don't move the boat until the ready state.

M1chaelM, 2022-04-06:
@alexperez33 Could you please post the command you are using to run the tasks when you see this error?

alexperez33, 2022-04-06:
@M1chaelM The error occurs whenever I launch ```vrx.launch``` using the urdf that was generated with the ```generate_wamv.launch``` script. This is due to the fact that the urdf that was generated with the ```generate_wamv.launch``` script does not contain the two joints specified above.

The script ```generate_wamv.launch``` does not allow us to send a parameter into the ```wamv_gazebo.urdf.xacro``` to override the default "locked" parameter (which is false). This "locked" parameter is in a xacro conditional which adds the above two joints.

All this to say, because my urdf does not contain these two joints, I receive the above errors and the boat does not stay locked during the initial state.

I run the generate_wamv.launch file as such:
```roslaunch vrx_gazebo generate_wamv.launch thruster_yaml:=<path_to_thruster_config> component_yaml:=<path_to_component_config> wamv_target:=<target_location_for_urdf_to_be_generated>```

The solution I provided in PR #446 allows a user to run the above command and add a parameter to turn the "locked" parameter true.

Another solution would be to just make the default parameter for "locked" in ```wamv_gazebo.urdf.xacro``` to true.

M1chaelM, 2022-04-07:
@alexperez33 Thanks for these details, and the PR. I'm going to look into this today.

I believe the error you are describing should be produced by a scoring plugin if it attempts to release a wamv that is not locked. Launching `vrx.launch` by itself should not cause any scoring plugin to run because there is no task associated with it by default. Are you running something like:
```
roslaunch vrx_gazebo vrx.launch world:=`pwd`/stationkeeping0.world urdf:=$WAMV
```

alexperez33, 2022-04-07:
I apologize, I forgot to mention that I am passing in a world as an argument. The error is produced by scoring_plugin.cc in the ```ReleaseVehicle``` function. It causes an error because it cannot find the joints.

M1chaelM, 2022-04-07:
Thanks @alexperez33 ! I was looking to see whether we might be able to get closer to the root of this problem. Since the need to lock and unlock the wamv is only relevant when the scoring plugin is running, it would be nice if the scoring plugins could manage both locking and unlocking, rather than depending on the wamv to be created a certain way. Unfortunately I think that is going to be too much to change right now. In the interim, the fix you propose looks good. I'll review the PR and we should also update the script `prepare_team_wamv.bash` in the `vrx-docker` repo so it automatically adds the locked options.

alexperez33, 2022-04-08:
Thanks! @M1chaelM. Sounds great!

closed 2022-04-09 by alexperez33