osrf/vrx #447: GPS and IMU related rostopics not visible

ashiriag, 2022-04-05:
Following up from email conversation.
When running the simulation through `roslaunch vrx_gazebo vrx.launch` and then listing the rostopics, none of the gps and imu related topics are visible.

M1chaelM, 2022-04-05:
Hi, since this was working for you on a different system, it sounds like you may be missing a dependency. Let's check whether there are any hints in the terminal output. Could you run
```
roslaunch vrx_gazebo vrx.launch verbose:=true
```
and post the output here?

ashiriag, 2022-04-07:
Thank you for this suggestion! By running that command I was able to see which ros dependency was missing and after installing it the issue is fixed. I had to run `sudo apt-get install ros-noetic-hector-gazebo-plugins`

closed 2022-04-07 by ashiriag